maxritter/diy-thermocam #5: choice of microcontroller

haraldg, 2017-08-31:
Hi,

this is a great project and I'm very likely to build a DIY-Thermocam myself.

However, I find it somewhat unfortunate, that the design is based around the TEENSY 3.6 instead of the 
[iMX233-OLinuXino-NANO](https://www.olimex.com/Products/OLinuXino/iMX233/iMX233-OLinuXino-NANO/open-source-hardware) which is cheaper, OSHW, industrial grade and already has many of the features on the board that DIY-Thermocam needs extra components for. Though admittedly getting things like the on board battery charger to work might require a non-trivial effort.

Anyway this issue is an attempt to find like-minded people who might join forces to adapt the design to the OLinuXino-NANO or solicit input on why this would be a bad idea.

maxritter, 2017-09-12:
The mentioned single board controller from Olimex is using Linux, whereas the Teensy 3.6 is a bare metal microcontroller. Of course you could also implement the Thermocam on a board running an operating system, like the Raspberry Pi Zero for example, but I decided to implement this project in bare metal C++ on an ARM Cortex M board.

haraldg, 2017-09-14:
The imx233 controller can definitely be programmed in bare metal C++, though maybe some libraries available for the Teensy might be missing. However having a full operating system available is actually the second reason why a Thermocam based on an olinuxino board would appeal to me.

I'm waiting for my order to arraive. After assembling and testing my own Thermocam I will decide how much effort I'm willing to put into a redesign. Of course any input anymone might have is still greatly appreciated.

haraldg, 2017-09-23:
As I'm still looking for people interested in porting the DIY-Thermocam to a OSHW single board computer, I'd appreciate if this issue could remain open to give it some visibility. Is there any reason to close it?

Nprod, 2019-02-01:
It's probably not what you have in mind, but it is possible to use just the FLIR Lepton breakout board directly on any Raspberry pi using the GPIO pins. If the olinuxino, or any other single-board computer of choice has I2C and SPI pins in its GPIO board, you should be able to get a Lepton running. I've successfully gotten it to run on a chinese Orange PI, the main downside being that it's running a full operating system so the whole setup isn't portable. 

Max's diy-thermocam is more expensive and more complicated to setup by comparison, but it has one big advantage that it's all integrated into a single unit. And when you're taking thermal images chances are you'll be out and about.

maxritter, 2019-02-01:
That's true :) There are various other projects and tutorial using a more advanced SOPC like the Raspberry for directly controlling and displaying the thermal image from the Lepton.

I think as you said, it depends on your use case and of course also the budget.

haraldg, 2019-02-02:
Yes. However I specifically suggested the olinuxino, because I believe such a setup would be as portable and the PCB interfacing the main board and the lepton would be almost trivial. (Ie making it easier to optimize for themal constraints, etc.)

The big part of such a project would be writing the firmware (probably mostly from scratch) as a linux application. My current estimate is, that developing a ThermocamV3 based on the olinuxino or other SBC would take me about 1-2 month full time work. - For something, that works. Doing it properly, ie. submitting drivers upstream to the mainline kernel, etc. would take even longer.

Unfortunately I don't have that much free time these days. Also the ThermocamV2 works too well for me, to put that much effort into something that would be slightly cheaper, a bit more flexible and OSHW.

However if somebody else is taking the initiative, I'd still be happy to help out with my skills. (Mostly kernel and embedded linux related.)

Nprod, 2019-02-02:
I don't know, sounds like a lot of work for something that's just 4€ cheaper. Maybe the added CPU power would make the image refresh rate quicker, but that's already limited to 9hz in the lepton itself...

closed 2021-03-10 by maxritter