maxritter/diy-thermocam #8: Calibrating problems

alexandreamato, 2017-09-03:
Hi,
I just built my own DIY-Thermocam. Congratulations on the project! If I were more careful I would have taken some photos of the built process to share with you and here. 
Anyway. I know it is working, at least a part of it, because I can start the calibration process, and I can move the arrows to put thermo image over photo image. But after that, calibrations asks to point to different heat objects slowly, but my calibration process never changes, stays at 0%. 
Also, I had to upload firmware version 2.40. I don't know why when uploading newer versions it says my teensy is not compatible.
One hint for everyone, I use Windows 10, and there was no way to upload firmware in COM10, but it was very easy with https://github.com/Koromix/tytools
Do anyone have any idea how to pass or bypass this calibration?
Thanks

nielsswinkels, 2017-09-04:
Did you see the Firmware update manual? In there at step 5 there is a link to how to change the comport on windows to com 10: http://plugable.com/2011/07/04/how-to-change-the-com-port-for-a-usb-serial-adapter-on-windows-7/

nielsswinkels, 2017-09-04:
Which teensy are you using?
And are you using the firmware updater?

alexandreamato, 2017-09-08:
I believe it is a 3.6 Teensy, my order says: 1568-1442-ND 32.50000 32.50, TEENSY 3.6, HTSUS: 8471.50.0150 ECCN: 3A991A2, LEAD: LEAD NA ROHS: ROHS NA, COUNTRY/ORIGIN: USA
I didn't use the firmware updater because tytools managed to update at least the 2.40 version without changing to COM10. Do you believe it is necessary? I will do it tomorrow!
Also, regarding calibration, can you imagine why I cannot progress? it stays in 0%?
I only had problems soldering the 2 smallest pins in the SD card, they were very difficult. But I believe they are not the most important ones?! Or am I wrong?

alexandreamato, 2017-09-09:
I did it, using the upgrade app and the com10 port I managed to upgrade to the latest firmware.
My only problem now is calibrating. It stays calibrating forever.
[image]
[image]
[image]
[image]
[image]

nielsswinkels, 2017-09-09:
I have to admit i don't know much, except that i recently put together one of these. 

I wonder if it could be an issue with the big cylindrical sensor? If i understand it correctly that is used to know actual temperature values from the values the heatcamera sensor says. Could you double check that is mounted correctly?

As for your photo, i can't see if the solder point is ok, it is to blurry for me. But you could always try to resolder it if you don't trust it?

alexandreamato, 2017-09-12:
I rechecked the big cylindrical sensor, it is positioned correctedly and soldered ok. Is it very fragile? Can it be broken? Do you have any idea how to test it?
Resoldering the location of the photo would be very difficult because of the location. But it is the SD card, so I studied the code, to see what happened when it stopped working. It doesn't yet access the SD card, so it is not the problem.
It seems it stops here:

`//If the temperature changes too much, do not take that measurement
//Get 100 different calibration samples
		while (counter < 100) {
			//Store time elapsed
			long timeElapsed = millis();

			//Repeat measurement as long as there is no new average
			do {
				//Safe delay for bad PCB routing
				delay(10);
				//Get temperatures
				lepton_getRawValues();
				//Calculate the average
				average = calcAverage();
			} while ((average == average_old) || (average == 0));

			//Store old average
			average_old = average;

			//Measure new value from MLX90614
			temp = mlx90614_getTemp();

			//If the temperature changes too much, do not take that measurement			
if (abs(temp - temp_old) < 10) {
				//Store values
				calLepton[counter] = average;
				calMLX90614[counter] = temp;

				//Find minimum and maximum value
				if (average > maxValue)
					maxValue = average;
				if (average < minValue)
					minValue = average;

				//Refresh status on screen in steps of 10, not for serial
				if (((counter % 10) == 0) && !serial) {
					char buffer[20];
					sprintf(buffer, "Status: %2d%%", counter);
// It arrives here at least once
					display_print(buffer, CENTER, 140);
				}

				//When doing this in serial mode, print counter
				if (serial)
					Serial.write(counter);

				//Raise counter
				counter++;
			}

			//Store old spot temperature
			temp_old = temp;

			//Wait at least 111ms between two measurements (9Hz)
			while ((millis() - timeElapsed) < 111);

			//If the user wants to abort and is not in first start or serial mode
			if (touch_touched() && !firstStart && !serial) {
				int pressedButton = buttons_checkButtons(true);
				//Abort
				if (pressedButton == 0)
					return;
			}
`
As "Status" remains at 0%, I imagine something is happening in this loop. I wanted to try the serial output, but the "FirstStart" routine runs the "calibrationProcess" with serial as false. I am really lost here.

maxritter, 2017-09-12:
Hey there,

sorry for my late reply, I was on vacation.

As the combined alignment screen is working, the FLIR Lepton sensor and visual camera are doing fine. The calibration screen utilizes the MLX90614 spot sensor for the first time.

Two things can go wrong here:
- You use the wrong version of the MLX90614. Only the BCF version is supported for the DIY-Thermocam V2
- You soldered the sensor in the wrong position. Check step 10 of the assembly guide and make sure the little indicator matches the mark on the PCB

Also check if there is a 3.3V voltage between VCC and GND of the MLX90614 and that SDA is connected to Teensy pin 18 and SCL to Teensy pin 19, with no connections to other pins. The datasheet contains the pin locations: https://github.com/maxritter/DIY-Thermocam/blob/master/Datasheets/MLX90614.pdf

Please tell me if that solved your problem.

Wishes,
Max

alexandreamato, 2017-09-12:
Hi Max, it is an honor to talk to you. Congratulations on your project. It is very interesting!!
I tried all the suggestions:
I believe the version is right, I bought it directly from the link in github, If starts with BCF, but the number following is 9696A. I attached a photo
[image]
The position is right:
[image]
Voltage OK
pin 18 and pin 19 seems ok. But I noticed is connected to other pins by the board.
[image]
[image]
[image]
[image]
the only piece I had to substitute was a missing 4.7k resistor that didn't came in the kit. I used a bigger one, but with the same resistance.

maxritter, 2017-09-13:
Thank you very much for your compliment :) 

From the pictures, everything looks solid! The spot sensor is also the right version.

So for the hardware side, I can only recommend you to resolder all pins and check if there are no hidden solder bridges, sometimes they are very small. Also check that the Lepton sensor is inserted tightly into the socket.

For the software side, you can try to reset the EEPROM so the spot sensor is flashed once again. Upload this sketch with the Teensy Loader, then reflash firmware 2.43: https://www.dropbox.com/s/jqcwolz5vajj0u5/eeprom_clear.ino.hex?dl=1

When in calibration mode, you have to move the device around to different temperatures, otherwise it will not make any progress. But a moving hand in front of the sensor should be enough to let it progress.

In case all that does not help, you can try to skip the calibration and directly proceed to live mode. Flash this sketch, I commented out the calibration part here: https://www.dropbox.com/s/jtglat0sg5msu17/NoCalibration.hex?dl=1

alexandreamato, 2017-09-13:
Unfortunatelly, resetting EEPROM did not work.
I double checked hardware soldering and bridges, resoldered the ones that looked ugly, cleaned the board. It also did not worked. 
Maybe my sensor is broken... I ordered a new one, but it will take a long time to arrive. 
So I installed the sketch with no calibration. And yes, it worked. At least now I can rule out a few other mistakes. Also when trying to run calibration from the menu, it locks the app, the abort button does not work. As the abort button did not exist in the firststart routine, I did not knew if it was a endless loop, or a problem that halted the software. Now I know it is halted. 
Anyway, I am happy with my partially working camera. I will need to compile my own firmware updates... but I can live with that.
Thanks

maxritter, 2017-09-13:
One step further in the right direction :) 
Maybe there is a weird software bug in the calibration function, that does not appear on my test devices here..

If you turn the spot sensor reading on in the Live Menu -> Live Disp. Options -> Spot On, do the temperature readings make sense? They come from the MLX90614 sensor.

alexandreamato, 2017-09-13:
It seems right... 
if MLX90614 is not working, value should be fixed?? because it changes regarding to what I am pointing
[image]
[image]
[image]
[image]

maxritter, 2017-09-13:
Thanks for the clarification!

From the hardware side, everything seems to be working, so it could be a software bug inside the calibration function.

Please flash this hex file with the Teensy Loader: https://www.dropbox.com/s/ifg5ketbz4ozoh4/CalibDebug.hex?dl=1

It contains debug outprints from 1 to 12 inside the calibration function, so I could see where it stops. You can use any serial terminal program like hTerm, connect to the serial port of the Thermocam with baudrate 115200 and send me the output when entering the calibration.

alexandreamato, 2017-09-13:
I need to go back to my lab to do that. Probably later today! Thanks

alexandreamato, 2017-09-14:
Here is the output:
[image]
[image]
[image]

maxritter, 2017-09-15:
Okay, so it got stuck in a while loop..

Please try that firmware file: https://www.dropbox.com/s/e6encj2gxblkum0/Firmware_V2_FixCalib.hex?dl=1

alexandreamato, 2017-09-16:
here it is. "Bad Calibration", but doesn't halt
[image]

maxritter, 2017-09-18:
Alright, so let's see what temperatures and raw values the device measures during calibration..

Please flash this file: https://www.dropbox.com/s/zqa7e7krh5r6io8/Firmware_V2_CalibOut.hex?dl=1

And send me the debug output as .txt / .csv file.
It prints three lines for each of the 100 measurements. First is id, second Lepton raw value and third MLX90614 absolute temperature. Either the raw value or abs. temp. must be wrong.

alexandreamato, 2017-09-21:
Hi, I had some trouble to upload the new firmware, but here it is:

7723
26.34
93
7723
26.14
94
7723
26.36
95
7723
26.36
96
7723
26.34
97
7723
26.34
98
7723
26.10
99
7723
26.36

alexandreamato, 2017-09-22:
a full one:

0
8149
36.88
1
8149
36.64
2
8149
36.76
3
8149
36.78
4
8149
36.36
5
8149
36.86
6
8149
36.94
7
8149
36.78
8
8149
36.60
9
8149
37.12
10
8149
37.12
11
8149
36.94
12
8149
36.94
13
8149
35.58
14
8149
37.60
15
8149
37.34
16
8149
37.24
17
8149
37.28
18
8149
37.60
19
8149
37.26
20
8149
37.26
21
8149
36.94
22
8149
36.94
23
8149
36.94
24
8149
36.60
25
8149
36.60
26
8149
37.02
27
8149
36.60
28
8149
36.62
29
8149
36.80
30
8149
36.84
31
8149
37.08
32
8149
36.72
33
8149
36.72
34
8149
36.60
35
8149
36.60
36
8149
37.10
37
8149
36.54
38
8149
36.44
39
8149
36.84
40
8149
36.54
41
8149
37.02
42
8149
36.86
43
8149
36.70
44
8149
36.54
45
8149
36.54
46
8149
36.68
47
8149
36.68
48
8149
36.78
49
8149
36.92
50
8149
36.92
51
8149
36.72
52
8149
34.98
53
8149
34.98
54
8149
28.16
55
8149
26.40
56
8149
26.76
57
8149
26.96
58
8149
26.90
59
8149
26.40
60
8149
27.84
61
8149
27.84
62
8149
27.14
63
8149
27.14
64
8149
27.52
65
8149
27.46
66
8149
27.66
67
8149
27.60
68
8149
27.14
69
8149
26.98
70
8149
27.42
71
8149
27.40
72
8149
26.66
73
8149
26.66
74
8149
36.30
75
8149
36.30
76
8149
35.36
77
8149
32.82
78
8149
32.82
79
8149
33.02
80
8149
34.42
81
8149
34.60
82
8149
35.14
83
8149
34.12
84
8149
34.60
85
8149
34.80
86
8149
34.80
87
8149
35.00
88
8149
35.00
89
8149
34.24
90
8149
34.42
91
8149
34.38
92
8149
34.54
93
8149
34.42
94
8149
34.58
95
8149
34.60
96
8149
34.60
97
8149
35.04
98
8149
35.04
99
8149
35.04

alexandreamato, 2017-09-22:
It seems Lepton raw value is wrong, not MLX90614 absolute temperature ??

maxritter, 2017-09-23:
Yes, it seems it is the Lepton in that case, because the raw value is always the same..

Please try that firmware: https://www.dropbox.com/s/fk6m2h6x4j9jcev/Firmware_V2_2.44B2.hex?dl=1
It has no debug outprints included any more, just see if you can get a decent calibration there.

nielsswinkels, 2017-09-23:
> fk6m2h6x4j9jcev/Firmware_V2_2.44B2.hex?dl=1

OK, let me get back to you

alexandreamato, 2017-09-23:
Did not work, See this video
https://vimeo.com/235138684/cab1ef6a1e

Maybe my lepton is different from yours? This was the only piece I already had from another project.
[image]
[image]

maxritter, 2017-09-26:
Yeah, I'm using the shuttered version of the Lepton for my projects here..
But it SHOULD work with the non-shuttered too.. Maybe the error is somewhere there, please try that approach: https://www.dropbox.com/s/0z0u3ikag7ykpz2/Firmware_V2_2.44B3.hex?dl=1
It removes the call to the Lepton shutter. On startup, there will be a model ID printed to the serial console, please send me that number as well.

alexandreamato, 2017-09-28:
Hi Max, I am out of town, as soon as I am back I will test.

alexandreamato, 2017-10-14:
Sorry for my delay. It seems to be working now.
At least the calibration finished without error.
This is the model ID in serial: 05-060340-

maxritter, 2017-10-15:
Thanks for your help!

I released firmware 2.44, that includes this bugfix: https://github.com/maxritter/DIY-Thermocam/tree/master/Firmware/Releases/2.44

alexandreamato, 2017-10-15:
Thanks a lot!
May I ask what is your final opinion about my problem?
What caused it?

maxritter, 2017-10-16:
The calibration method missdetected the Lepton2 non-shuttered as a Lepton2 shuttered.
At the beginning of the process, it tried to run a shutter command, which caused the raw sensor values to freeze afterwards.

closed 2017-10-24 by maxritter