maxritter/diy-thermocam #11: Data doesn't get transfered to PC

JustaCopy, 2017-10-21:
Hi,

i can connect my DIY-Thermocam (TC) with ThermalDataView and ThermoVision, but the programs don't recieve any data from the device. (No captured images, no live-stream)

I can charge the TC and update the firmware. Do you have any suggestion where i should start looking for?

maxritter, 2017-10-21:
What operating system are you using? For Windows, you can download the serial driver here: https://www.pjrc.com/teensy/serial_install.exe , for Linux here: https://www.pjrc.com/teensy/49-teensy.rules
Have you updated your device to the newest firmware 2.44?
Have you changed the comport to COM10?
Is there an error message when you try to f.ex. use the ThermalLiveViewer?

JustaCopy, 2017-10-22:
Thanks for the fast reply.
I use Windows 10. Driver you describe is installed. 
Yes, i use the newest firmware 2.44
Yes, i tried to change the comport to COM 10.
ThermoVision gives the error "DIY_noDataRecievedRestarts==5->Stream OFF"
Now i tried ThermalLiveViewer. There i can see the thermal image. So, no error there.

maxritter, 2017-10-22:
For issues concerning the ThermoVision SW, please contact the author directly with the given error message: joe-c-henkel@web.de

JustaCopy, 2017-10-23:
I'll ask him by chance. Your ThermalLiveViewer works pretty well, i guess i'll stick with it for now.
One last question, is it intended, that windows doesn't see the SD-Card?

maxritter, 2017-10-24:
The DIY-Thermocam V2 unfortunately does not support mass storage mounting for the SD card. This has to do with the internal wiring of the SD SPI bus. So the only solution is to use an external card reader.

closed 2017-10-24 by maxritter