maxritter/diy-thermocam #12: Error when plugging in SD-card

thorro7, 2017-10-23:
Hi,
whenever i plug in the SD- card (got one with 32 gb) the screen freezes and sometimes weird graphical bugs appear. Have you ever heard of this error? Or might the reason be the 32 gb instead of the proposed 8 gb? 

Thanks for your awsome project and support. Wishes

maxritter, 2017-10-23:
I think your card somehow is not compatible with the SDFat library, thats why it freezes..
Please try out a smaller card (4GB / 8GB), the one from Sandisk mentioned in the part list works for sure. Also make sure you formatted it with FAT / FAT32.

thorro7, 2017-10-25:
Hi, 
thanks for your fast reply. While formatting then card with FAT32 didnt solved the problem, a SD- card with 4 GB didnt lead to any issues. I still dont know WHY the error occured with the 32 GB card but ill just go with the smaller one.

closed 2017-10-25 by maxritter