maxritter/diy-thermocam #13: battery size doesn't fit

haraldg, 2017-11-22:
Hi,

finally got my DIY-Thermocam kit from groupgets. There are some minor annoyances, like I got two tripods but miss some other parts and the import tax was way havier then expected. However the real issue is not their fault: The battery doesn't fit between the display connector and the push button!

I just checked. They sent me this battery: https://www.sparkfun.com/products/13855, which is actually linked from the DIY-Thermocam readme. However this battery has the following dimensions: 5.6mm x 49.2mm x 68.8mm, while the README states the following max. dimens.: 60mm (w) x 55mm (l) x 6.5mm (h)

I think the link in the README should be removed and maybe groupgets notified, that they need to update their kit.

maxritter, 2017-11-22:
Hey,

thanks for your feedback and sorry for the inconvenience.

Can you please post an image showing the battery in place? I suspect the length of the sparkfun battery is too high?

Wishes,
Max

haraldg, 2017-11-23:
Sorry, can't take a photo at the moment. Yes, the problem is the dimension of 68.8mm of the sparkfun battery. 60mm is the maximum - as is actually stated in the README correctly.

maxritter, 2017-11-23:
Thanks for the observation.

I have to deeply apologize for this mistake. I added this new battery type and didn't check the specifications carefully enough.

It is fixed now on the website and I also forwarded this to GroupGets. I hope not too many users are affected by it. 

Concerning your particular case, please also get in touch with GroupGets, as I do not have any replacement parts for the battery here.

haraldg, 2017-11-24:
Don't feel to bad about this. Such mistakes are bound to happen and in the grand scheme of things buying a second battery is only a minor annoyance. Actually I'm very happy to have a thermal imager where I can improve the firmware whenever needed and it will cost me only twice the prize of a cheap commercial solution in the end.

For now I taped the too large battery to the exterior of the case and ran some first tests. I guess I should open new tickets for the issues I find. However some of the things I want to discuss are really just related to how to do high quality measurements - information that would be interesting to many users I guess. If there was some kind of forum, where such posts would stay easily accessible, that would be very useful IMO.

Anyway, closing this issue now.

closed 2017-11-24 by haraldg