maxritter/diy-thermocam #14: firmware stops responding sometimes

haraldg, 2017-11-24:
It seems there is some condition, where the firmware gets stuck: The display freezes and the firmware doesn't react to any touch or button events. Turning the camera off and on fixes the problem.

I can't exactly reproduce it, but it happend to me three times now over a period of 1-2 hours. It always happens in the process of taking/saving an image. I have the gut feeling, that it might be related to bouncing of the button or the exact time or duration the button is pressed.

Sorry, I don't know how to dig into this further ATM. But I will try to get a feeling of how I need to press the button to trigger the crash.

maxritter, 2017-12-02:
Thanks for the info.

Right now I am not able to reproduce this issue on my test devices here. If you could nail it down a little bit further, that would be very helpful :)

haraldg, 2017-12-02:
I think that the freezes mostly happen when I push the button a bit harder but rarely when pushing it very gently. So looks like a problem with bouncing to me. If you have a better button than what groupgets sent me, then maybe you have no chance for reproducing ...

maxritter, 2018-04-17:
I increased the debouncing time in FW version 2.46. Please check it out and report back the results :)

FuwoWolf, 2019-10-05:
Well, I seem to have to Problem as well, but it seems not only to happen while taking images, but also when leaving the menu or at random in live view.

I'm on the up to date on 2.48

closed 2021-03-10 by maxritter