maxritter/diy-thermocam #15: thermal analysis software for alternative OS

haraldg, 2017-11-27:
Hi,

it seems only the live viewer software is platform independent. All the other applications for working with thermal images are only available for windows it seems. Also I couldn't find the source code for any of them, so it seems porting any of them to linux is not a viaable option either.

Before is start writing something from scratch: Does anybody know of a free software program, that would be a good starting point for working with thermal raw data files?

BTW: As a first step I intend to add support for raw *.dat files to the traditional file(1) command.

maxritter, 2017-12-02:
Thats true. But you can contact the author of the ThermoVision software (joe@joe-c.de), and ask for the sources. Maybe its also possible to run it with a Windows compatible runtime environment like Wine.

However, it would be very interesting to see more interest on an open-source, multi-plattform thermal analysis software. Unfortunately, I do not have the time to implement it on my own at the moment..

haraldg, 2017-12-02:
If he is willing to release his code as free software, that would be very nice - no matter how much use his Windows code would be for me personally. If it's just me getting the code to look at, then I'd rather not waste my time with it.

However I guess it is time to update this ticket with my own findings: I'm currently looking into the calibration issue #16 and testing various open tools in the process.

There is https://gmic.eu/ - it caught my attention because it claims to work well with multispectral images. Indeed it is a powerful and useful tool for this task, but it's own script language is the most horrible programming language I have ever gotton something productive done with.

So at the moment I'm looking into the various python libraries, in case they are a viable alternative.

I expect to post a review article about DIY-Thermocam capabilities for precize measurements in a few days, which at the same time will also be a HOWTO for processing and analyzing the raw data with free software.

haraldg, 2017-12-14:
I have written a minimal set of hopefully os independent tools here:

https://github.com/haraldg/thermocam-tools

maxritter, 2017-12-14:
This set is very helpful, it's great to see some more software tools for this device!

closed 2021-03-10 by maxritter