maxritter/diy-thermocam #16: poor calibration performance

haraldg, 2017-11-27:
Hi,

the automatic calibration procedure is quite nice - at least it is much simpler than any manual calibration would be. However results are somewhat underwhelming: Most of the time my DIY-thermocam shows temperatures about 3-6K too low. I think the problem mainly comes from the spot sensor: It shows quite reasonable values around room temperature, but at colder temperatures the values become quite too low: For example today the temperature outside was clearly above 0°C (air and any surfaced I can think of), but the spot sensor showed temperatures around -8°C.

Is there any way to recalibrate the spot sensor itself, to see if this yields better calibration of thermal images too?

Im also thinking about just taking raw data with the thermocam and do any calibration and analysis on the PC later. But this is made more difficult by #15.

haraldg, 2017-12-14:
I have written a short article about thermocam calibration:

https://github.com/haraldg/thermocam-tools/blob/master/manual-calibration.md

maxritter, 2017-12-14:
Thanks for this nice article! I added it to the documents so other users can better find it, hope this is okay for you :)

haraldg, 2017-12-14:
Of course it is okay for me. It was written for the benefit of other people after all ...

However I might update it to improve my rather poor english and maybe also with additional results in the future. So maybe linking to it would be easier to maintain than including it.

I am rather suprised about the poor spot sensor performance - looks pretty much useless to me. I wonder whether there is some way to improve this on the firmware side.

haraldg, 2018-01-17:
So I looked into the spot sensor (calibration) problem a bit more and it turns out my initial diagnosis was wrong: It is NOT the ambient temperature that matters, but just the uptime of the thermocam: After switching the thermocam on, the spot sensor starts at reasonable values, but then drifts down to lower values.

I guess this is caused by the camera warming up (it gets noticably warm even on the outside after about half an hour of operation). I notice that the used sensor should be resistant to internal temperature gradients to mitigate this very issue. However, unless the sensor I got is broken in some subtle way, it seems the thermal regime inside the thermocam is just too much.

I guess the only real fix for this is to redesign the thermocam with heat distribution in mind. This probably is not going to happen for quite a while, so I'm looking into work arounds.

Solution 1: Disable the spot sensor completely, just using manual calibration. This will need a change to the firmware, I think.

Solution 2: Reduce heat production inside the thermocam. This would probably only reduce the issue with the spot sensor, not mitigate it. OTOH it would help with other things (drift of the lepton, battery life) as well, so might still be worthwhile. I don't know how the components contribute to heat production, but I guess a huge part comes from the microcontroller.

There are a number of ways, how I think the microcontroller could be made more energy efficient.
* make it sleep instead of busy-looping in some places of the code
* reduce the work load - for example I rarely use the optical camera, yet it seems it can't be actually disabled
* reduce the clock speed

I guess all these ideas require modifying the firmware. So my first task is to look into how it can be compiled on linux. Any help with that or with the issue in general would be very appreciated - I'm new to arduino, so will need a while to get up to speed.

maxritter, 2018-01-18:
Thanks for this further investigation.

I already assumed something like this, as I also made kind of the same observations. The internal heatup seems to be too much, so that the MLX90614 can not compensate for it.

Maybe it's possible to find some compensation formula for this self-heating. If I know what exactly to implement in the firmware, I can easily do that :)

haraldg, 2018-01-20:
About compensation: The spot sensor drift is surprisingly repeatable. Even if I change parameters like adding/removing the battery (should be a huge heat buffer), the curve (drift over time) stays pretty much the same in my tests. However, I think a time-based compensation won't work, because I have to reboot the camera when I'm hit by #14.

However there are at least two internal temperature sensors available: The internal temperature on the MLX90614 and the die-temperature of the MK66. Maybe something can be done with either of those...

Also it seems that the drift of the spot sensor is pretty much synchronized with the drift of the lepton sensor itself. On one hand this makes the spot sensor pretty much a no-op, on the other hand experimenting with the built-in temperature compensation of the lepton becomes much more interesting. (I believe it is disabled by default on the non-radiometric versions, but the feature is there.) To that end I'd like to experiment with register settings on the lepton. To make this feasable, the lepton CCI would need to be forwarded via the thermocam serial interface.

About the warmup itself: As I have only one thermocam, I can't take a thermal image of the thermocam itself. However I found the following forum post / image: https://forum.pjrc.com/threads/45012-how-warm-the-teensy-3-6-gets?p=146777&viewfull=1#post146777

Looks to me, like the main heat comes from the µC and two rather inefficient regulators. I guess all the power for the lepton and the display also comes from these regulators, so they might be really hot in the thermocam. I still intend to experiment with lower clock speeds (to simulate the effect of more efficient code), but I suspect the teensy 3.6 might not have been a good choice for such a project - at least not without adding external regulators.

wiiverueckt, 2018-03-26:
Is it possible to port the code to rapberry pi zero?
The CPU is much faster and also the price is much lower.
I think its also programmable in arduino if you want.
Maybe even the videoout could be implemented.
Does the drift stop after a while or is it drifting all the time?

haraldg, 2018-03-28:
Hi @wiiverueckt,

regarding using other CPUs there is issue #5 - I'm still interested in this.

The drift is high at the beginning and reduces over time as the camera approaches thermal equillibrium. However this takes rather long, probably the battery is depleted at the same time. I usually use the thermocam for about an hour in a row and this is not enough.

Actually the lepton sensors have a mode (not used by the thermocam firmware ATM) where the sensor tries to compensate for changes in sensor temperature. It is on my todo-list to experiment with this mode - could be that in this mode the lepton alone can produce better results than the calibration sensor. Unfortunately I have little time to work on this.

blotske, 2018-04-03:
@wiiverueckt it would be harder (not impossible) to do without a realtime processor.  That is why its very common for a realtime microcontroller to be attached directly to the sensor/device and that connected to a higher level OS and processor.

haraldg, 2018-04-18:
Ah, sorry I forgot to report some of my on results in this issue: I compiled my own build of the firmware some time ago at 120 MHz clock speed IIRC. The result was a mixed bag: The drift is slower at lower clock speed, so it kind of confirms our theory. But it is still strong enough to make the spot sensor quite useless, so I didn't run enough tests to get a detailed comparison. I suspect at 120MHz the main heat source actually is the regulators supplying the display with power, but since displayless mode is not useful for me, I did not experiments to check this theory.

Instead I'm meaning to look into the temperature compensation abilities of the lepton itself. However this requires some significant changes to the firmware, maybe adding some kind of debug mode to make experimenting with lepton settings easier. I don't have time to work on such a project in the near future, so I can only encourage you and others to look into this and offer to help from the sidelines.

closed 2021-03-10 by maxritter