maxritter/diy-thermocam #18: USB connection stuck at "lepton_getRawValues()" function

reanovich, 2017-12-21:
I only have Teensy 3.6 and Lepton 2 (shutter) at the moment and connect them with cables. Every time I try to run the firmware, it will always stuck at  "lepton_getRawValues()". This problem occurs because teensy never got 60 lines from lepton. It always loop at line 0 because the error value is count up to 255. I have tried with 2 pcs Lepton 2 (shutter) and 1 Lepton 2(non shutter) and the problem still exist. This is how I connect teensy with lepton :

- pin 11 to MOSI
- pin 12 to MISO
- pin 14 to CLK
- pin 15 to CS
- pin 18 to SDA
- pin 19 to SCL

I powered them from USB so I connect this two pin
- pin GND (Teensy) to GND (Lepton)
- pin Vin (Teensy) to Vin (Lepton)

I already test the Lepton module with my own code to read the package (using SPI and SPI1) and I can get number 0 - 60 from the first two bytes (ID bytes). So there should be no problem with my Lepton module.

Any idea why this happens?

maxritter, 2017-12-21:
Can you post the full code you are using to test this setup?

If you connect CLK to pin 14, you also need to issue startAltClockline() in addition to SPI.beginTransaction(SPISettings(20000000, MSBFIRST, SPI_MODE1)) and digitalWrite(pin_lepton_cs, LOW) to begin a transaction with the Lepton2.

What is the LeptonReadError retVal, when you try to read a package?

reanovich, 2017-12-22:
Hi Max, I'm using this code to test my Lepton module
<details><summary>I'm gonna use this fancy spoiler because the code is too long</summary><p>

```Arduino
#include "Wire.h"
#include "SPI.h"

#define USE_SPI1
/* Wiring
- SPI1
	-- CS	pin 6
	-- MOSI	pin 21
	-- MISO	pin 5
	-- SCK	pin 20
- SPI
	-- CS	pin 10
	-- MOSI	pin 11
	-- MISO	pin 12
	-- SCK	pin 13
*/

#ifdef USE_SPI1
#define pin_lepton_cs 6
#define pin_lepton_mosi (21)
#define pin_lepton_miso (5)
#define pin_lepton_sck  (20)
#else
#define pin_lepton_cs   (10)
#endif

void setup() {
	// put your setup code here, to run once:
	Wire.begin();
	Serial.begin(115200);

	Serial.println("Lepton module test");

#ifdef USE_SPI1
	// Initiate lepton SPI
	Serial.println("Init SPI1");
	pinMode(pin_lepton_cs, OUTPUT);
	digitalWriteFast(pin_lepton_cs, HIGH);
	SPI1.setMISO(pin_lepton_miso);
	SPI1.setMOSI(pin_lepton_mosi);
	SPI1.setSCK(pin_lepton_sck);
	SPI1.begin();
#else
	// Initiate lepton SPI
	Serial.println("Init SPI");
	pinMode(pin_lepton_cs, OUTPUT);
	digitalWriteFast(pin_lepton_cs, HIGH);
	SPI.begin();
#endif
}

void loop() {
	// put your main code here, to run repeatedly:
	int line = 0;
	byte leptonFrame[164];
	byte imageFrame[60 * 164];
	while (line != 60)
	{
#ifdef USE_SPI1
		// Start transaction
		SPI1.beginTransaction(SPISettings(20000000, MSBFIRST, SPI_MODE1));

		// Initiate CS pin
		digitalWriteFast(pin_lepton_cs, LOW);

		// Receive one frame over SPI
		for (int i = 0; i < (164 / 2); i++){
			// Read value via SPI
			leptonFrame[2 * i] = SPI1.transfer(0x00);
			leptonFrame[2 * i + 1] = SPI1.transfer(0x00);
		}

		// Disable CS pin
		digitalWriteFast(pin_lepton_cs, HIGH);

		// Stop transaction
		SPI1.endTransaction();
#else
		// Start transaction
		SPI.beginTransaction(SPISettings(20000000, MSBFIRST, SPI_MODE1));

		// Initiate CS pin
		digitalWriteFast(pin_lepton_cs, LOW);

		// Receive one frame over SPI
		for (int i = 0; i < (164 / 2); i++){
			// Read value via SPI
			leptonFrame[2 * i] = SPI.transfer(0x00);
			leptonFrame[2 * i + 1] = SPI.transfer(0x00);
		}

		// Disable CS pin
		digitalWriteFast(pin_lepton_cs, HIGH);

		// Stop transaction
		SPI.endTransaction();
#endif

		if ((leptonFrame[0] & 0x0F) != 0x0F){
			if (leptonFrame[1] == line){
				for (int i = 0; i < 164; i++){
					// Save frame
					imageFrame[(line * 164) + i] = leptonFrame[i];
				}
				line++;
			}
		}
	}

	// Print lepton frame ID
	for (size_t i = 0; i < 60; i++)
	{
		Serial.printf("%02X\t", imageFrame[(164 * i) + 1]);
		if (i % 10 == 9)
		{
			Serial.println();
		}
	}
	Serial.println();

	while (1);
}
```

</p></details>

The output should be like this
```
Lepton module test
Init SPI1
00	01	02	03	04	05	06	07	08	09
0A	0B	0C	0D	0E	0F	10	11	12	13
14	15	16	17	18	19	1A	1B	1C	1D
1E	1F	20	21	22	23	24	25	26	27
28	29	2A	2B	2C	2D	2E	2F	30	31
32	33	34	35	36	37	38	39	3A	3B
```

Somehow when I tried my code with CS at pin 15 and SCK at pin 14, it just hung up and never print any result (after init SPI) so I failed to replicate the same connection as DIY-Thermocam to make the lepton work.

the retVal value are 0,1, and sometimes 3

maxritter, 2018-04-17:
I am closing this issue now, as it's not directly related to the DIY-Thermocam.

closed 2018-04-17 by maxritter