maxritter/diy-thermocam #19: RAM Requirements for ThermoCam

xmdfb, 2018-01-09:
Hi, I know you are using the Teensy 3.6. I was wondering how much RAM memory I would need if I were to choose my own micro controller. How much did you need to process the 160 x 120 thermal images.

maxritter, 2018-01-10:
The DIY-Thermocam firmware uses two buffers to store the images:

//320x240 buffer (Teensy 3.6 only)
unsigned short* bigBuffer;
//160x120 buffer
unsigned short* smallBuffer;

The larger buffer can contain an image with 320x240, so that it can be displayed in the full-screen resolution of the LCD display. This is used to upscale the thermal images with some filtering and therefore provide smoother images. But it is not required for a proper functionality and can be enabled or disabled with the HQ mode setting inside the menu. In addition to that, it is only available on the Thermocam V2 using a Teensy 3.6 MCU with 256K RAM. This array uses 320*240*2 (16-Bit) = 153.6 kByte

The smaller buffer is required to store a thermal image. It has a memory requirement of 160*120*2 = 38.4 kByte. The Teensy 3.2 used in V1 of the DIY-Thermocam offers 64kB of RAM. This can be considered as the absolute minimum, if all the other GUI features of the Thermocam (Load / Save / Processing) are used. 

So for a short answer, a minimum RAM of 64kB is required, if GUI features are used and HQ mode is disabled. When HQ mode is enabled, 256kB are necessary.

Hope that helps :)

haraldg, 2018-01-17:
@xmdfb Out of curiosity: Which microcontroller are you considering? Why?

I note that you seem interessted in smaller microcontrollers while I'd like to see bigger ones (see #5), but I'm still interested what criteria other people have.

Meins321, 2019-05-08:
So the lepton gives 320x240x16=1228800bits
1228800bits/8= 153600 bytes
so (kilo=1000) gives 153,6 kilobytes... or 150kb if you devide by 1024 :-)

closed 2021-03-10 by maxritter