maxritter/diy-thermocam #20: compiling the firmware

haraldg, 2018-01-21:
I'm trying to compile the firmware myself, but I have some trouble understanding how things should fit together.

I got the Arduino IDE (1.8.5) and Teensyduino (1.41) installed, which was quite non-trivial as I'm on arm64 but only 32-bit arm binaries are provided. Anyway, it seems to work now.

But getting Visual Studio as is suggested in your README for arm64 linux seems not viable. So I tried building the firmware directly from Arduino IDE. After renaming the Folder "Source" into "DIY-Thermocam" I could open the sketch and the compile phase was successful. However the linker produced lots of errors about unresolved symbols.

Basically the problem seems to be, that the "external libraries" aren't actually external, but are copied into the source tree. The compiler finds and includes all header files, so there is no problem. However the *.cpp files of the libraries never get compiled, so the linker has nothing to link actually.

A few questions:
1) Before I start messing with things: Do you have any recommendation on how to compile your code without Visual Studio?
2) Since I guess it would be the most straight forward solution: Why are the external libraries copied into the project? Would I run into any trouble, if I just switched back to using the official libraries? Would such a change be acceptable for you upstream?
3) I guess, I could write a Makefile for the libraries. But how would I then use this "manually" compiled library to build the sketch?

haraldg, 2018-02-01:
Meanwhile I successfully compiled the Firmware on arm64 linux with the arduino IDE only.

Basically the following steps where necessary:

* Add symlinks from $(HOME)/Arduino/libraries, to the external libraries that can't be installed from the IDE. These are ADC, I2C and Metro.
* Install all the other used libraries from within the IDE.
* Move tjpgd.c and tjpgd.h into their own folder in the external libraries directory and add a symlink for that as well.
* Probably optional: Comment (or remove) the #include directives for the external libraries in the main sketch file and replace them with proper #include directives for system libraries.
* Symlink (or copy) the remaining *.cpp and *.c files the the toplevel source directory, so Arduino IDE sees them. Also include the referenced *.h files - not many! Maybe the best way for this would be to just move these files into their own DiyThermocamHelpers "external" library.

That's it. Not all the difficult after all.

I guess it would make sense to provide a pull request for some of these changes, to keep differences between my branches and the master branch to a minimum. Which of these changes would be acceptable to you? Also I won't be able to test compilation on Windows / with Visual Studio and I can't update proprietary Visual Studio files, if I move some source files around. I'd need you cooperation on that part.

What do you think?

maxritter, 2018-04-17:
I will provide an extra repo for the firmware soon, that includes a refactored version of the source code as well as a tutorial that will be based on Eclipse. This way it can be also used on other operating systems ;)

maxritter, 2018-04-17:
The updated firmware repo is now online!

haraldg, 2018-04-18:
Thanks, that looks very promising! Having an extra repository for just the firmware does make my life a lot easier. I hopefully have time to test this (and the debounce improvement) on the weekend.

I think in the long run it might be useful to have a way of fully automatic building the firmware for continous integration testing. Not much of a concern now, but once people start to contribute lots of pull requests, this would be nice to have.

haraldg, 2018-04-21:
Compiling the new firmware on linux fails because of case sensitive filenames! I can prepare a PR, but do you want all filenames lower case (like they are used in the code) or should the code be changed to mixed case filenames?

maxritter, 2018-04-22:
Thanks for the info. I changed all module filenames to lowercase now.

haraldg, 2018-05-04:
Thanks, I successfully compiled the firmware now on linux / arm64. I still used a Arduino IDE only approach (ie without eclipse), but no changes to the source files where necessary this time (only renaming main.cpp to src.ino) so I can now easily keep track of firmware updates without having to merge local changes anymore.

I haven't tried it yet, but I expect it will be easy now to setup automatic building with arduino-builder from the command line. As far as I'm concerned, this issue can be closed now.

Thanks!

closed 2018-05-06 by maxritter