maxritter/diy-thermocam #21: Microcontroller

saammbautista, 2018-01-31:
Is it possible to use arduino nano/pro mini instead of teensy? Thank you!

saammbautista, 2018-01-31:
And is it possible for the videos/images to be transmitted wirelessly instead of in a sd card and has a touch display? I am planning to use it for a thermal camera equipped drone. Thank you!

blotske, 2018-04-03:
@saammbautista The teensy is way more powerful than an arduino.  It might be possible but not nearly as performant.  Also, the project already has a touchscreen.

I just finished building my camera a couple weeks ago.  My project is to get it flying for search and rescue.  I'm planning on processing the images from the camera on the airframe.  I haven't figured out a image/video downlink quite yet.  Lots of options but most have limited range.  

I'm planning on creating a new board that will allow the sensors to be gimbaled with the teensy and power management rigid. Of course anything I create will be contributed back to this project.

maxritter, 2018-04-17:
The Arduino nano or pro mini does not have the required hardware capabilities to display thermal and visual images in real-time on the 320x240 display. Also it will be very slow, the Teensy 3.6 for example can be overclocked to up to 240MHz.

You can use the analog video signal of the video output module together with a 5.8GHz FPV transmitter to transmit the thermal feed wirelessly to a ground station.

closed 2018-04-17 by maxritter