maxritter/diy-thermocam #23: Lepton 3.5

Loremwriter, 2018-03-29:
Hello,

Just a little question from germany. :)
I want to start to build this project, but im just wondering, if I can order the Lepton 3.5 Sensor instead of the lepton 3.0. 
Is the 3.5 radiometric or do I have to expect any obstacles while building your thermocam with a 3.5 instead of 3.0?

https://www.digikey.de/scripts/DkSearch/dksus.dll?Detail&itemSeq=256344870&uq=636579089465335088

Best regards

maxritter, 2018-03-31:
The Lepton3.5 is pretty new, it was just released last week.
It is fully radiometric like the Lepton2.5, so you do not need the MLX90614 spot sensor.

I will hopefully get my Lepton3.5 by the end of next week, so I can make the firmware compatible with it. On the hardware side, you don't have to worry about any changes.

Wishes,
Max

Loremwriter, 2018-03-31:
That's great!
Yes its quite new, but could you maybe post a reply, if it works with Laptop 3.5. 
At this moment I would order my parts to build the cam. Or do you think I can order it anyway, because the firmware will be compatible then? :)

Thanks. :)

maxritter, 2018-03-31:
You can go ahead and buy it, I will make the firmware compatible next weekend if my sensor is already there until then ;)

ulab, 2018-03-31:
I was thinking about upgrading my cam to the 3.5 since I broke the shutter when taking apart my One for the sensor. Does it hurt if the spot sensor is soldered in when switching the module?

maxritter, 2018-03-31:
Nope, the spot sensor is just not used when a radiometric Lepton is detected.