maxritter/diy-thermocam #25: Fahrenheit not displaying properly in temp bar

JSField, 2018-04-09:
I have a Lepton 3 with the spot sensor, and when I turn on the unit in °F mode, after the warmup, the numbers in the bar are wrong. They go from low negative numbers to extremely high positive numbers. If I change to °C, the numbers are still wrong, until I reboot the camera, then they are ok. If I switch to °F, the bar numbers are OK, unless I try to use Manual, try to adjust the min or max, the + and - buttons change the values by large steps. Something appears to be wrong with °F conversion.

JSField, 2018-04-09:
Heres what it looks like when in °F mode:

[image]

maxritter, 2018-04-17:
Thanks for the observation.

There was indeed a mistake in the fahrenheit conversion, that caused the mentioned bugs.
It's fixed now and will be implemented in the next FW release.

maxritter, 2018-04-17:
Should be fixed in FW version 2.46. If not, please reopen this issue.

closed 2018-04-17 by maxritter