maxritter/diy-thermocam #26: ThermoVision not connecting to camera

JSField, 2018-04-09:
I've tried using the ThermoVision software from Joe-C, but cannot connect to the camera, at least not getting a live stream. I have firmware v2.44, and have tried ThermoVision v1.7 and 1.8 also. I am able to get a live stream from Thermal Live Viewer, on COM10 though. I've contacted Joe-C and sent him some troubleshooting info, but I was wondering if anyone else is having this problem?

maxritter, 2018-04-17:
Yes, I am having the same issue with the latest version of the ThermoVision software and live mode..
Have you received any response from Joe-C so far?

JSField, 2018-04-17:
Yes, Joe-C emailed me back and asked for more info on the issue, but I haven't heard back in 2 weeks. I'm going to post in the ThermoVision forum post on eevblog.com and see if I get a response. Maybe he didn't get my last email.

Loremwriter, 2019-10-03:
Any news about it?

How did you make the live stream work?

JSField, 2019-10-11:
I did get a response from Joe-C, last year, and he sent me a couple of versions to try, but I didn't get it to work consistently. I haven't really tried it since last year, and version 1.9.0, but I see he has a version 1.10.0, but haven't tried it yet. I will give it a try soon.

Vital1984, 2020-03-14:
1.10 dont work for me