maxritter/diy-thermocam #27: Updating from FW1.24

gumbymoegoe, 2018-04-15:
[image]

Hi I'm trying to get our cam up to the latest FW but am having issues with the Updater.
When I hit connect the cam says that its connected and turns its screen off but the updater gets stuck on the hardware version detector part.
is there an older version of the updater available so that I could maybe bring it up to speed in stages?

Thanks

maxritter, 2018-04-15:
You can continue flashing without an activate connection, this will only check battery status and current firmware.
If the DIY-Thermocam Updater does not work, you can also use the Teensy Loader. Hope that helps!

gumbymoegoe, 2018-04-16:
Hi Max,
Ended up going the Teensy route as the Updater was not completing the flash.( after pressing the flash button a pop up box asking to confirm comes up however after confirmation it still displays awaiting user input below flash button).

anyhow Teensy worked and it is now all updated 
Thanks for the help

closed 2018-04-16 by gumbymoegoe