maxritter/diy-thermocam #33: Cannot connect to camera as a mass storage device after firmware update

jesass, 2018-05-14:
I have a version 1 DIY Thermocam that I have been using that has been running version 1.25 firmware.  All functions supported by the 1.25 firmware worked as expected including mounting the camera as a mass storage device to transfer images to a P.C.

After I upgraded the firmware to version 2.46, I cannot mount the camera as a mass storage device.  If I go into the camera menu and select the USB File Transfer icon and then select the YES button, the camera briefly displays the "Disconnect USB cable to return"  message and then the camera reboots.

Is mounting the camera as a mass storage device supported in the 2.46 firmware for version 1 hardware?

System Info:
DIY Thermocam version 1 hardware flashed from version 1.25 firmware to version 2.46 firmware
Windows 10
Camera connected on COM5

maxritter, 2018-05-16:
Yes, the newest firmware should also support mass storage mode on the DIY-Thermocam V1. But I have to admit I didn't test it in version 2.46. Can you flash 2.45 or a prior version and check if it works there?

jesass, 2018-05-16:
Are there old firmwares archived here?  I am having trouble finding any firmware except the current 2.46 version.

maxritter, 2018-06-05:
You can find previous firmware releases here: http://diy-thermocam.bplaced.net/Firmware/

jesass, 2018-06-08:
Thanks for the link to the previous firmwares.  I did some testing with my version 1 camera and saw slightly different results as the firmware progressed from v.1.25 to v.2.46.  The results are below.

From v.1.25 to v.2.39 - Normal operation was observed.  Normal operation is defined as the following:
- The USB icon is pressed from the main menu.
- A message pops up that says "Disconnect USB cable to return" is displayed.
- Windows USB disconnect sound is heard followed by the Windows USB connect sound.
- A drive opens in Windows Explorer displaying the contents of the onboard SD card.  The contents of the drive can be accessed.
- When the USB cable is unplugged, the camera reboots.

Starting with the v.2.40 firmware to the v.2.44 firmware, a slightly different behavior was observed:
- The USB icon is pressed from the main menu.
- A message pops up that says "Disconnect USB cable to return" is displayed.
- Windows USB disconnect sound is heard.
- Nothing happens
- If the USB cable is disconnected and then reconnected,  the Windows USB disconnect sound is heard when the cable is disconnected and then the Windows USB connect sound is heard and then the contents of the SD card is displayed in Windows Explorer.  The contents of the drive can be accessed.  The message "Disconnect USB cable to return" is still displayed on the camera display.
- When the USB cable is unplugged, the camera reboots.

With the v.2.45 firmware, the normal behavior (v.1.25 to v.2.39) was observed.

With the v.2.46 firmware the following behavior was observed:
- The USB icon is pressed from the main menu.
- A message pops up that says "Disconnect USB cable to return" is displayed.
- Windows USB disconnect sound is heard.
- The camera reboots.

I took a look at the change log for 2.45 to 2.46 and did not see anything that would point to the change in behavior between v.2.45 and v.2.46.

I am not asking for a fix to this since I realize that I have an old device and the newer device does not seem to have problems.  I am posting this so that others that may have version 1 camera have a resoure to help solve problems after firmware upgrades.

Thanks again for all your help with this issue.  With v2.45 firmware, I have a functional camera again !

maxritter, 2018-06-17:
Thanks for those detailed observations!

I had a look what could be the issue with 2.46 and the MSD functionality, but didn't find a quick fix.. Hopefully I will have some more time to investigate this soon, and hope you can use 2.45 without any big issues in the meantime :)

jesass, 2018-06-24:
The v.2.45 firmware works fine for what I am doing and it is a vast improvement from the v.1.25 that I was using!

If you are going to do some troubleshooting on this issue, I am willing to test any potential fixes that you come up with.

closed 2021-03-10 by maxritter