maxritter/diy-thermocam #35: SD Card not seen

stevefleischer, 2018-05-19:
I am currently building my second Thermocam, the first one was successful, but this time I have hit a problem.
The device boots up and I go through the setup screens without a problem. Unfortunately, the start up process stops with the message "Checking SD card..." and then "Please insert SD card". However, the SD card is inserted into the card reader properly. It is the one I used to test the first camera with and it worked then, I was able to save various thermal images to it.

My questions are:
1) is a faulty card reader connection the only cause of this error (given the card has been tested in another Thermocam)
2) is it possible to bypass the SD card detection part of the boot up process so that I can test the actual FLIR is working? I don't (at the moment) need to save any images.

nacerazel, 2018-05-19:
Hello , i m working on my first thermocam, my touchscreen doesn t work, please can i contact you ? Or can you give me the steps to install the thermocam?

stevefleischer, 2018-05-20:
Hi,
I'm not sure if this is relevant to your problem but this was my experience. After I built my first camera and switched it on the screen was white. There was no boot up message or anything. Just a white screen. I checked the voltages on all the components and everything seemed OK so I was puzzled about what could be wrong. One of the last instructions in the assembly guide said to flash the firmware. I didn't think this was necessary because I had done this right at the beginning, before I soldered the Teensy onto the PCB. However, I didn't know what else to do at this point so I thought I would flash it again. The problem was that I couldn't access the button on the Teensy (because it was between the microcontroller and the PCB). I read some posts on this page and found one that mentioned accessing the Teensy using software called tytools (which was written by Koromix on GitHub). I downloaded it and ran it (on Windows 10) and it displayed the details of the Teensy board. I used the Upload button to flash the firmware again and after that the boot screen came up and the Thermocam started working. 

So, in short, my camera had a white screen because I needed to flash the firmware a second time. I did this using tytools. I hope this helps.
[image]

nacerazel, 2018-05-20:
i had the same probleme before uploading the firmware, my toouchscreen shows me a message (first Booting and then a quick message which show me the devices that are ok and the ones failed, ( touch screen and spot senssor (i use lepton 2,5) ) and after this a message " please touch screen for your first use " is written but it doesn t respond,

stevefleischer, 2018-05-21:
I'll have to write this build off as a failure. Now it freezes on the boot message screen and then the LED starts to flicker, screen goes white. There seems to have been a problem with the battery right from the start but at least I got to the "Insert SD card" screen. Now I can't even get that. Although the battery reads 3.98V (supposedly 3.7V - 2000mAh) there was always a bootup message to say that the battery needed charging which I ignored. Furthermore, the Pololu 5V voltage regulator gets extremely hot - painfully so - and that can't be right.  As I am out of ideas now (I replaced the Pololu but it made no difference) I will have to admit defeat.

maxritter, 2018-06-05:
Seems like there is a short somewhere on the PCB, which can be very hard to identify.. Where did you get the PCB from, or did you produce it by yourself?

stevefleischer, 2018-06-05:
Thanks for your reply. I got the PCB from Smart Prototyping using the Gerber file your provide. They look very good and I don't think it is at fault. Initially I got to the boot up screen but it couldn't detect the SD card. I think I might have damaged something when I re-soldered the card reader tested the voltages on the board. Anyway, a bit pricey to buy the components and start again - at least for now. I'll shelve the project for a later date. My first build worked perfectly (this was my second build) so thanks for sharing your great project.

closed 2018-06-05 by stevefleischer