maxritter/diy-thermocam #36: Wrong Arducam delivered

Loremwriter, 2018-05-21:
Hello,

I wanted to build my own thermocam, but I got stuck because of the wrong parts. I used the links in the description, but I got some new version of the "Arducam 2MP Camera Shield“. 
I used this link: https://www.ebay.com/itm/Arducam-Mini-module-Camera-Shield-w-2-MP-OV2640-for-Arduino-UNO-Mega2560-board-/281639729684

Unfortunately I bought it on eBay, but this camera doesn’t fit in the case. 
I guess we need the version like IMG-1, but I got the „new“ version as seen on IMG-2. 

IMG1:
[image]

IMG2:
[image]

Could you please tell me, how I can use the Version I got, or how I could get an "older" version of this camera?
Guess GroupGets still delivers the correct version, so where do they buy their stuff?

Can't imagine, that I'm the only with this issue. :)

Best regards :)

blotske, 2018-05-21:
I used the new version successfully.  I don't have a case for it yet though.  It will post the design files once completed.

Loremwriter, 2018-05-21:
@blotske : So you built it with V2 and it works?
Thought the position of the lens would have a big impact to the results. I had the idea to use a lens from another cam, instead of this one. 

Seems to be a 1/4" diagonal 60 degree lens.

If anyone has any idea to this topic... Tell me. 
And it would be nice, if you could post your design files, even thought my enclosures are cut already.

blotske, 2018-05-21:
The hybrid view is slightly off but not by much.

Loremwriter, 2018-05-26:
Update: I just drilled a hole with 15mm diameter into the front panel, works fine. :)

closed 2018-05-26 by Loremwriter