maxritter/diy-thermocam #39: Camera stopped working: can't identify the issue

JeremyG, 2018-05-30:
Hi,

I managed to build the cam, flash the firmware, and start the cam. I was on the calibration screen when I turned it off, intending to do the calibration on the following day. Then the next morning, the cam didn't start. I first thought it was a battery issue, but I tested it, it charged OK, until 4.10 / 4.20V approx., and then the light turned green.

However, the cam doesn't start anymore. The issue seems to come from the Teensy: can't see the orange control led anymore, and the tension between Vb and Gnd is around 1V, nothing more.

It may also be coming from the coin cell battery: I thought it may be drained and changed it. The new one half-drained quite quickly, and it didn't change the teensy issue.

What do you think my issue would be? Is the Teensy dead because of a short circuit or something I might have done involuntary? Could it be coming from the coin cell battery? Could I have damaged the PCB in some way?

I have an extra Teensy 3.6. Do you think I can go ahead and change it right now, or is there a way to identify the source of the issue first?

It's already a lot of questions so I leave it there :). Any help appreciated, thanks!

maxritter, 2018-06-05:
The coin cell battery is only responsible for the integrated real-time-clock, so it does not influence the boot of the device. 

You should be able to measure around 5V between VIN and GND on the Teensy. If this is not the case, there is something wrong with the Pololu step-up booster. Also, the Teensy shoukd generate 3.3V between the 3V3 pin and GND. 

If the supply voltages are stable and the backlight work, you might want to check the connections between the Teensy and the Display. In case the hardware is OK, try reflashing the firmware.

JeremyG, 2018-07-11:
It appears that I burned the microcontroller, probably after a bad handling of the battery causing a short-circuit. I changed the faulty components and re-built the cam, it's working now.

closed 2018-07-11 by JeremyG