maxritter/diy-thermocam #44: Video out module: image too big for the SD card?

JeremyG, 2018-08-14:
I'm trying to flash the video module image to the SD card on Windows 10 with Etcher but it appears that the image is too big for the SD card: image appears as being as large as 7.95 GB, while the SD card only has 7.82 GB available.
Am I the only one experiencing this? 
Not a critical issue as I could switch for a bigger SD card, but if it's a recurring issue it might be worth mentioning it on readme file?
I'll try on macOS and report back here.

JeremyG, 2018-08-14:
Same issue on macOS.

maxritter, 2018-08-16:
The size of the SD card can vary, depending on the manufacturer and model.

If you use the 8GB model from Sandisk, the image should fit. For other cards, following this tutorial may help: https://www.raspberrypi.org/forums/viewtopic.php?t=62233

JeremyG, 2018-08-20:
My bad! I didn't pay enough attention that a Sandisk SD Card was specifically needed and bought another brand. The link to the article on ebay doesn't work here however (France).
I fixed it by using a 16 Gb Sandisk SD card bought locally. A bit overkill maybe but I'll be able to capture tons of videos and images then :)

closed 2018-08-20 by JeremyG