maxritter/diy-thermocam #45: Lepton 3.5 - high temperatures (>200°C)

markusr, 2018-08-24:
First, thank you very much this wonderful project. We bought a DIY-Thermocam V2 (Groupget Kit) with a Lepton 3.5. The hardware is working fine.
However, we have problems measuring temperature over 200°C. It seems that the values (on display and also in the .dat-file) are capped at 181.92 °C. 

The Lepton 3.5 can support up to 400 - 450 °C in low gain mode. Is this a hardware or a firmware problem?

markusr, 2018-08-24:
I did some digging into the firmware source code. It seems that currently the setting of the gain mode (SYS Gain Mode) is not implemented. 

The high gain mode (LEP_SYS_GAIN_MODE_HIGH) is the default setting which explains that the Lepton 3.5 does not show values up to 450 °C.

maxritter, 2018-08-25:
Yes thats right, changing the gain mode is currently not implemented in the firmware. Unfortunately, I do not have time for this project at the moment, so any help from the community is highly anticipated.
If someone finds time for that, please do a pull request in the firmware repo.

markusr, 2018-08-28:
@maxritter 

I have added some code for setting the SYS gain mode. See my commit:
https://github.com/markusr/diythermocam_firmware/commit/5827fbf218dd78e0443fef044529e1c375aec7c0

During init (for the lepton 3.5) the gain mode is set to "low gain". I will see if I can also add some GUI code for switching during usage.

markusr, 2018-08-29:
After some testing I have the next problem. The low gain mode works as expected. However, now the values are capped at 382.2 °C.

The maximum value of the 16-bit is 65535. This is (with scale factor 0.01) 655.35 Kelvin. This equals to 655.35 - 273.15 = 382.2 °C

I will see if and how this can be solved.

markusr, 2018-08-29:
I found a solution by setting T-linear Resolution parameter to 0.1. 

I have updated my branch.
https://github.com/markusr/diythermocam_firmware/commits/master 
(Includes a re-factored version of the lepton code)

The highest measured temperature was 459.85 °C. (The real temperature was ~650 °C)