maxritter/diy-thermocam #49: Slicing software having trouble to properly scale .stl files for enclosure

sergek, 2018-10-04:
I know these are contributed files, but the uploaded .stl files for 3d printing have some units issue.
Slicing software thinks that the models are tiny.
Does anyone know if it's possible to fix this without downloading the 'inventor' and re-exporting those .stl's?

maxritter, 2018-10-05:
Have you tried out the single parts from here: https://github.com/maxritter/DIY-Thermocam/tree/master/Enclosure/ArducamV1_Radiometric/3D%20Print ? Do they also have the mentioned issue?

sergek, 2018-10-05:
Hello, yes, I'm talking specifically about .stl files from that folder.

maxritter, 2018-10-05:
They were not created by myself, but by Llorenç Macià. I will notify him about this Github issue. Maybe some automatic STL fixer like this one helps? https://tools3d.azurewebsites.net

LenzoM, 2018-10-07:
Hello Sergek,

I made it by inventor, you are right.

Few days ago, when I send it to another friend of mine he appointed the same as you do.
I don't know exactly how to solve this, but I can redraw it. During this week I will try to sort it out.

But, if I am not wrong, the scale is 1/10. Hence, scale the dimensions and you get the piece as it should be. Make sure that the walls are thick as 2,4 mm the siders and 3 mm the closers.

sergek, 2018-10-07:
Hi Llorenç,
If I'm right about the nature of the issue, there is no need to redraw anything, it's a matter of exporting of the model from Inventor to cad file formats.

I will try your suggestions, thank you!

LenzoM, 2018-10-07:
Thank you very much. So I will re-export it properly and I will pull a request with the update!

maxritter, 2018-10-10:
The parts should be fixed in the latest commit: https://github.com/maxritter/DIY-Thermocam/tree/master/Enclosure/ArducamV1_Radiometric/3D%20Print

Please report back if this solves your problem :)

sergek, 2018-10-10:
Thank you, it's much better but still not quite there yet unfortunately .

I've test printed the bottom.stl and it came out 1/4th smaller than the original.
Please look at the picture attached.
The wide side of a black acrylic part is 100mm, the printed part is 75mm wide approx
And I think this is not the problem with export but with the model itself but I may be wrong.

 
[image]

LenzoM, 2018-10-15:
Dear Sergek,

I am having some troubles because I don't know the exact measures I have to achieve.

So it means that I am scaling the parts I have drawed. So I explain my procedure.

1. Here there is a raw file of svg. This file has what I assume is right about the sizes and shapes.
2. I can't download it (with windows 10) so I copy all the content to a txt and then i save it in svg format
3. From inkscape to autocad I have to save it in pdf 
4. For a reason I can't find out (although I have checked all about units and scales) from 101.405 mm that  measures in inkscape, when I import the pdf file to autocad it measures 3.15 mm
5. I scale it all to the right measures I could get from inkscape (wich is the 101.405 mm the largest)
6. From autocad to inventor and I extrude it 3 mm
7. I save it in stl making sure I save it in high quality (over 1mb per file more or less) and in mm as well

But I repeat my problem: I do not know exactly what are the measures of the holes, the radius, etc.
So maybe Max can help us out.

BTW, I am thinking about redrawing it in a 3dprinting capable, which it means to reinforce some little parts, like the holes next to the borders, which, after I see the result of the printing, it is quite week and it needs to be way stronger

maxritter, 2018-10-25:
The 3d printing files have been updated: https://github.com/maxritter/DIY-Thermocam/tree/master/Enclosure/Radiometric/3D%20Print

sergek, 2018-10-31:
Hi, it took me some time to get back and try these updated files and I've tried only the front and back covers, but I'm happy to say that after some very light treatment the parts fit very nice.
Thank you very much for making and tuning these 3d printing files!