maxritter/diy-thermocam #50: Calibration stuck at 0%

leonardp, 2018-10-22:
Hi,

after the initial setup (setting date etc.) the screen shows the calibation screen stuck at 0%.
We waited for about 1minute or more but it did not change in any way.

Any hints on how to get more debug output out of the firmware or any other ideas how to tackle
this problem would be appreciated.

Thanks
Leo

maxritter, 2018-10-25:
Dear Leo,

are you using a non-radiometric version of the Lepton with the external spot sensor, or a radiometric Lepton? 

If you are stuck in the calibration screen, it seems like the connection to the lepton and/or the spot sensor is not working. Please use a multimeter and check the lines between those components and the Teensy.

You could also flash this file (https://www.dropbox.com/s/jtglat0sg5msu17/NoCalibration.hex?dl=1) with the Teensy Loader (https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Firmware%20Updater/Teensy%20Loader) and afterwards the newest firmware again to skip the calibration process. But if the Lepton or the spot sensor is not working, you may be stuck in the boot screen then.

Hope that helps!

Wishes,
Max

Meins321, 2019-05-13:
In my case the Lepton sensor was lose in the 90° pinheader and missing one solderspot (oops)

It was "calibrating" after i checked the lepton and the lasersensor

leonardp, 2020-09-10:
Turned out to be a defective MLX90614-BCF Spot Sensor.
Having Sanity checks for that would be nice to have in the firmware (if it isn't already implemented).

closed 2020-09-10 by leonardp