maxritter/diy-thermocam #51: Silkscreen mirrored in gerber file

Nprod, 2018-11-06:
Hi there, i'm preparing to have a few PCBs made of the latest version (2.3) and the silkscreen text for the major components like the camera, lepton breakout etc. is mirrored in the gerber preview. Is this normal? Both the viewer at http://www.gerber-viewer.com/ and the one at http://jlcpcb.com shows reversed text for the bottom layer while the top layer is fine.

maxritter, 2018-11-13:
Looks pretty normal to me:

[image]

Also when I manufactured them, I didn't notice any mirrored silkscreen..

Nprod, 2018-11-14:
This is how it looks in JLCPCB's viewer, comparing it to your picture it looks like the entire board is mirrored on the bottom view.
[image]

I guess it's a quirk of the website, i wanted to make sure it's correct since it's not the only online gerber viewer that showed it like this.

closed 2019-01-29 by Nprod