maxritter/diy-thermocam #53: Problems FIRST firmware installation

LenzoM, 2018-12-28:
Hello,

I can't install the firmware for the fist time in the camera. I have just assembled it all and so I want to flash the firmware.

With the updater from the software folder, and the last HEX files from the firware section. I connect the camera to the computer ( I have installed the DRIVE.exe) and I get these messages.

How do you see them?

[image]

[image]

[image]

LenzoM, 2019-03-07:
Hello Max!

Thank you for your reply :)

I, after some troubles, finally managed to connect the camera to de computer and finally I could reboot the hex file, until i get the message, from the teenesy program of OK. But, after that, my screen gets frozen in this picture and it doesn't seem likely to move on.

[image]

I wish you could help me!

stephen94000, 2019-03-25:
Hello, I'm currently facing the same problem as you do. Have you managed to bypass this screen ?

stephen94000, 2019-03-25:
I think this is somehow related to the hardware detection of the components in the device?

maxritter, 2019-03-25:
Dear @LenzoM , dear @stephen94000 ,

sorry for my late reply, I was on vacation. Stephen is right, this is most likely related to one or some of the hardware components not working / not connected properly. 

Please use a multimeter and check all the connection to the sensors (SPI & I2C Lepton, I2C MLX90614 for non-radiometric, SPI & I2C camera). You could use those two documents for that purpose: https://github.com/maxritter/DIY-Thermocam/blob/master/Documents/Pinmap_2.1.pdf & https://github.com/maxritter/DIY-Thermocam/blob/master/Documents/Schematic_2.3.pdf. Also check if the supply voltage for all components is correct and the Lipo is fully charged.

If the problem still persists, you could install the Eclipse IDE, load the firmware project there and insert Serial Outprints at different stages of the startup procedure. This way you can identify where the procedure hangs up. Setting up the IDE is described here: https://github.com/maxritter/diythermocam_firmware

Let me know if this helps :) 

Best,
Max

Meins321, 2019-05-08:
Mybe this helps and you can add it to the reository for people without eagle!

Is shows the traces from one spot to another
[PCB 2.3.pdf](https://github.com/maxritter/DIY-Thermocam/files/3156544/PCB.2.3.pdf)

LenzoM, 2021-02-25:
Thank you for your kind help! It works now fine!! 
I have another issue, but i open now another topic

closed 2021-02-25 by LenzoM