maxritter/diy-thermocam #60: DIY ThermoCam USB tethered use

DaleStokes, 2019-02-19:
Wondering if anyone running Firmware 2.47 has been able to connect to Windows 10 via USB and any software (Thermal Live viewer, ThermoVision or, Thermal Data Viewer).   My recently fabricated unit (excellent job Max!!!) works great autonomously. However I am unable to activate live viewing/streaming on any software. The computer does recognize the ThermoCam on COM 10 serial USB and the camera is detected by software but the software immediately crashes -- it looks like the streamed values are out of range producing div /0 errors for instance.   This kind of USB tethered operation would be extremely helpful.

maxritter, 2019-03-25:
Dear @DaleStokes,

thanks for sharing those observations. I just tested the software with the latest firmware 2.47. 

The Thermal Live Viewer 2.03 works fine on my Windows 10 x64 PC. Maybe you are having problems with the compiled version of it. In that case, install Python (f.ex. Anaconda) with all import dependencies (f. ex. pip install pygame) and launch the ThermalViewer.py directly (python ThermalViewer.py in shell).

The Thermal Analysis Software does indeed not stream the thermal images, it prints out "DIY_noDataRecivedRestarts==5->Stream OFF". You might contact the autor Joe-C and ask him for help: joe@joe-c.de.

Best,
Max

DaleStokes, 2019-03-25:
Thanks Max!

I'll give it a try....  If anyone else has had success/problems with tethered use, please feel pass on your advice. 
cheers,

Dale

closed 2021-03-10 by maxritter