maxritter/diy-thermocam #61: Initial setup shows "Lepton2.5" when 3.5 is installed + note and question

BruXy, 2019-02-21:
Hi, I have join latest GroupGet campaing and finally assembled the camera. It is great project, congrats!

**1. I am just curious, initial setup showed "Lepton2.5", but I should have 3.5. Just for clarification, this is the picture of what I get:**

http://bruxy.regnet.cz/hamradio/diy_thermalget.jpeg

I have compared it with Digikey picture and it is Lepton 3.5 (not sure 100%, but looks same):

https://www.digikey.com/product-detail/en/flir-lepton/500-0726-01/500-0726-01-ND/6163867

I have dig little bit around source code and looks like, it is a feature:

https://github.com/maxritter/diythermocam_firmware/blob/master/src/gui/firststart.cpp#L302

**2. I had also problem with initial uploading of firmware in Linux**

I have used:

`./teensy_loader_cli -v --mcu=TEENSY36 -w  Firmware_V2.hex`

It was waiting for Teensy to appear, but to be able to upload firmware, I needed to press the reset button, a.k.a shorting P and G, which I have found here: https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Firmware%20Updater

**3. Visual and thermal image position**

Should both images be perfectly aligned or it is more for just rough comparison?

My camera shows this: http://bruxy.regnet.cz/hamradio/thermalget_test.jpg

I am courious if it should be just rough or should I spend more time with alignint of both cameras together? But I was quite precise when puting all spacers under modules.

Thank you for answers and clarification!

clesys, 2019-03-03:
Hi BruXy,
maybe off topic, bt if you installed a radiometric Sensor like Lepton 2.5 or 3.5, you don't need the single-point infrared sensor.
Compare "AssemblyGuide" Step10

maxritter, 2019-03-25:
Dear @BruXy ,

let me answer your points:

1.: You are right, it's a typo in the screen message displayed. The device should have found out already you are using the leptonVersion_3_5_shutter variant of the sensor, if the thermal image is displayed correctly. @clesys is also right about the fact that you don't need the MLX90614 when using the Lepton2.5 or Lepton3.5, but it shouldn't affect the detection of the sensor. It's just not used by the firmware then.

2.: Thanks for the information. When flashing the firmware for the first time, depening on the OS it can happen that you need to short those two pins. I am glad it worked then.

3.: Due to the small paralaxe between camera and Lepton, the alignment depends on the distance to the scene / target. I am having different alignmnents saved for standard distance scenarios, like 1m, 5m, 10m. I find the combined mode helpful for a rough clue what objects in the normal image correspond to which in the thermal image, so I wouldn't spend too much time fine-tuning here.

Best,
Max

closed 2019-11-27 by BruXy