maxritter/diy-thermocam #63: ArduCAM-M-2MP Plus SPI Initialize issue

SenexCrenshaw, 2019-03-20:
I am having a really hard time for some reason finding a non-plus module. I received a ArduCAM-M-2MP Plus module and got spi errors when the firmware starts up. I even put just a demo app on the thermocam using pin 10 as the CS pin. Both ways i can see SPI initialization errors. Everything else works great. So i decided to try another camera and i got the same plus model... Same issue. I used my meter to make sure pin 10 does connect to the cs pin on the camera. So at this point I am not sure if its something wrong with my soldering which i did it twice or the board I received has something wrong with it..

Any ideas??? 

Awesome project btw. It works great but I would love to have it 100% functional.

kaimart, 2019-03-21:
Same here. Watterot only sells the plus model, old ones are out. I didn’t look for logs yet, on startup the software says „init failed“ for the visual camera. 
Great project, however! I really appreciated the work on it with this very well documented order an build process. Thanks!

maxritter, 2019-03-25:
I was not aware there is a plus model, but seems like they changed something in the driver code there so it's not working with my implementation any more.

From the code changes in the Arducam repo alone I can't see whats the difference: https://github.com/ArduCAM/Arduino/tree/master/ArduCAM. The #define for OV2640_MINI_2MP_PLUS seems to behave the same than for the OV2640_MINI_2MP. 

If someone can send me his module, I can fire up the IDE and see whats wrong. Please drop me an email to mail@maxritter.net for the shipping address.

Best,
Max

maxritter, 2019-04-27:
Firmware 2.48 should provide support for the ArduCAM-M-2MP Plus: https://github.com/maxritter/diythermocam_firmware/releases/tag/2.48

closed 2019-04-27 by maxritter