maxritter/diy-thermocam #64: How to order the board?

Thanh-Binh, 2019-04-01:
Hi,
could you please inform me how to order the board or complette system?
Best thanks

WoodsideOak, 2019-04-05:
When I built the camera, I had to order 10 printed circuit boards.  They are for the V2 Thermocam, Rev 2.3 boards.

I only used one board and will sell the others at cost (1GBP each) plus postage.

[image]
[image]

KrisJot, 2019-04-19:
Hi,

I'm new on github.

@WoodsideOak what copper thickness did You choose, 1oz,? I'm interested with ordering one from You.

WoodsideOak, 2019-04-22:
@KrisJot Yes 1oz copper.  Board spec as follows:
    - Material: FR-4 TG130
    - Layer: 2
    - Dimension (Length): 89
    - Dimension (Width): 68
    - Solder Mask Color: Green
    - PCB Thickness (mm): 1.6
    - Copper Thickness: 1 oz
    - Surface Finish: HASL (Hot Air Solder Leveling)
    - Min. Tracking / Spacing: 6 mil
    - Min. Hole Size: 0.3 mm
    - Fiducial Mark: Yes
    - Polishing Technique: Polishing

KrisJot, 2019-04-23:
@WoodsideOak, thats great, would like to buy two pcb from You.

WoodsideOak, 2019-04-24:
@KrisJot No problem.
The boards cost me 1GBP each so I will sell for that price. 
Please send me a private message with your address and I will calculate postage to you.

FuwoWolf, 2019-07-21:
Hi @WoodsideOak , are there still boards left?

WoodsideOak, 2019-07-21:
@FuwoWolf, yes I have boards left.

I've sent you a PM with postage costs.

hasulhas, 2019-10-16:
Hi @WoodsideOak , i want to buy them all

WoodsideOak, 2019-10-16:
@hasulhas - No problem, i have 8 boards left and will sell them at £1 each.
Please email me your address and I will calculate the postage.

turbopropello, 2019-10-16:
Would love one of them too

jonaswinterhalder, 2019-10-17:
@WoodsideOak Would like to buy one too. I will write you.

jbiz111, 2020-02-19:
Hi, id like to buy 2 or 3 if still available.

Fischgeburt, 2020-12-01:
Has anyone 1-2 PCBs which aren't needed anymore?

WoodsideOak, 2020-12-04:
> 
> 
> Has anyone 1-2 PCBs which aren't needed anymore?
> 
> @WoodsideOak Maybe?

Yes, I have 5 boards left and will sell them at £1 each.
Please email me your address and I will calculate the postage.

Fischgeburt, 2020-12-06:
@WoodsideOak 

Sent you an email.

User35123, 2021-01-06:
@WoodsideOak 

is still one available? I would like to get one.
Thank you!

WoodsideOak, 2021-01-12:
Hi,

 

Yes, I have two boards left.  I will sell them at cost price of 1GBP each, plus postage.

 

If you send me your address I will let you know what the postage would be.

 

Regards

Malcolm

closed 2021-03-10 by maxritter