maxritter/diy-thermocam #66: TP4057 Charging Module replacement

KrisJot, 2019-04-21:
Hi,
 
can be used TP4056 as replacement for TP4057 from part list?

maxritter, 2019-04-27:
It does not exactly fit the holes on the PCB, but from the charging perspective, it should work. However, you need to solder cables to the LEDs on the module and remove them, if you want to use the charging light functionality.

KrisJot, 2019-04-30:
Thank you for reply. Will try and let you know about the result.

Meins321, 2019-05-08:
> 
> 
> It does not exactly fit the holes on the PCB, but from the charging perspective, it should work. However, you need to solder cables to the LEDs on the module and remove them, if you want to use the charging light functionality.

Maybe it is the better option?

I did buy the TP4057 off ebay with blue pcb, but i got the green pcb and it doesnt work. My lipo did not charge and the whole board just heated up like warm toast.
the teensy is dead now :-(

The lepton still works as i tested it with another socket

So don't use the green TP4057 which is delivered from ebay from many sellers

KrisJot, 2019-05-23:
Hi, TP4056 is charging properly (blue PCB). Lipo charged to 4.2V.

closed 2021-03-10 by maxritter

rossoe, 2023-01-25:
@Meins321 
Is this the appropriate wiring for the TP4056 ?
[image]

rossoe, 2023-11-28:
See updates on this [post](https://github.com/maxritter/diy-thermocam/issues/103) for final tested wiring solution.