maxritter/diy-thermocam #69: Issue with initial Thermocam USB connection

FOC1902, 2019-05-08:
Dear Max, dear community,

first of all, thanks to that great project. It must have been a lot of research, time and work. Great job!
I was building the Thermocam the last couple of days and stumbled across an issue while connecting it to the computer the first time. Once I connect the Cam, the charging LED lights up. If I switch on the cam, the LED on the arduino starts blinking and the display turns on showing a white screen. Every ~30seconds, I can also hear the thermal image sensor clicking. And thats it, Window does not recognize the cam at all. 
Not sure how to proceed further - is that behavior normal or do I have an error somewhere in HW?

Thanks,
Bastian

Meins321, 2019-05-09:
Maybe your battery is depleted and the usb charging port of you pc or laptop cant handle charging+startup?
Do you have the same behaviour on another pc/usb port? Did you charge the cam completely before it was connected to the usb port?

FOC1902, 2019-05-10:
Hi,

yes I charged the battery until the LED turned green. I also tried different USB ports and even another computer. Does anyone know if the behavior which I described above (display lights up with white background, clicking of the thermal image sensor etc.) is the standard behavior? 
I appreciate any idea... thanks in advance

Meins321, 2019-05-13:
Sounds like this:
https://github.com/maxritter/DIY-Thermocam/issues/65

"the screen lights up I can hear (and see) the thermal sensor shutter closing and reopening"

maybe your lipo is bad or a solder connection, if it was mine i would open it and put a multimeter on the battery to see if it sagging on startup

You said: "Window does not recognize the cam at all." <- Teensy 3.6 micro usb connection broken? After all it is just mico usb, take out the plastic plate ontop of it an check maybe with another calbe/charger combo?

FOC1902, 2019-05-16:
Hi all,
I finally found the error. Seems like I had a bad welding connection somewhere at the switches, I re-welded them and now it works fine. 
Cheers

closed 2019-05-16 by FOC1902