maxritter/diy-thermocam #74: Video Output Module - Image leads to 404

Loremwriter, 2019-08-17:
See headline. 
Cant download the firmware image for the module.

Gerrit28282, 2019-08-30:
I partly solved the issue. You can find the firmware here: 
https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Video%20Module/2.02 
Just follow the readme an power on the pi.

For me the script does not autostart, I have to manually start it from the console via: python2 __init__.py

Depending where that file is located on your SD Card you first have to navigate to that place. For me it is: cd /boot.

Note that the python script is written for python 2.7, some functions like map() and chr(CMD_FRAME_DISPLAY) dont work as supposed unter python 3. 

Nevertheless, the issue with the thermal live viewer still exists.

Loremwriter, 2019-10-01:
Hi Gerrit, 
i just have to copy the "__init__.py" and "config.txt" onto the SD card?
Its the only content, that i can find in the 2.02 zip file. 

So i dont have a boot directory or something like that. 
Or do i have to flash a raspbian version to the sd card?

Thanks in advance! :)

Concerning the thermal live viewer: Im currently in contact with the developer of the app. We are working on a solution to solve the Lepton_Connect Error.

Gerrit28282, 2019-10-01:
You should have installed a recent Raspbian distribution on your SD card. Easiest way to do that is with the [NOOBS](https://www.raspberrypi.org/downloads/noobs/) installer. 

After that you should find a boot partition on your SD card where you can put both the config and the "__init__.py" file. The init file is basically the firmware which you can manipulate to fit your own needs. 

Would be great if you could share your solutions regarding the live viewer if it works. 

Cheers, 
Gerrit

Loremwriter, 2019-10-01:
Great, it works! :)

I will share the solution if it works. Probably it takes some more time and will end in an updated version by joe-c.

ncbelov, 2021-01-18:
Hello! I managed to defeat the autostart of the _init_.py file?