maxritter/diy-thermocam #75: Last Time Buy Date	09/30/2019 LEPTON BREAKOUT BOARD EOL or Obsolete Part!!

Meins321, 2019-08-28:
@maxritter Got an E-Mail from Digikey:

Part Affected
--
Manufacturer | FLIR LEPTON
Description | LEPTON BREAKOUT BOARD PCB ASSY
Manufacturer Part Number | 250-0587-00
Digi-Key Part Number | 250-0587-00-ND

[Link](https://www.digikey.com/classic/ObsoletePart.aspx?dkcid=3163&WT.z_email=obsolete&pn=250-0587-00-ND)
Status | End Of Life
Last Time Buy Date | 09/30/2019
Substitutes | Please click here

blackdot, 2020-04-07:
Hello, this substitute has different pinout.
Did anyone managed to make everything work?

jonaswinterhalder, 2020-04-07:
Hi blackdot, im working on it. I just received the Breakout Board v2.

VoodooBane, 2020-05-01:
> 
> 
> Hi blackdot, im working on it. I just received the Breakout Board v2.

Any update on this? Would this [https://store.groupgets.com/products/purethermal-breakout-board] work as a drop in replacement for 250-0587-00?

Let me know I ordered the groupget "PureThermal Breakout Board" just in case

jonaswinterhalder, 2020-05-04:
> > Hi blackdot, im working on it. I just received the Breakout Board v2.
> 
> Any update on this? Would this [https://store.groupgets.com/products/purethermal-breakout-board] work as a drop in replacement for 250-0587-00?
> 
> Let me know I ordered the groupget "PureThermal Breakout Board" just in case

Unfortunately I got a broken camera and could not continue testing. The camera is currently on its way to America. I'm sorry for the bad news.

Pesylizium, 2020-06-08:
Does anyone already have new information about the Breakout Board V2?

mdjonni, 2020-06-23:
> Does anyone already have new information about the Breakout Board V2?

Anyone got it working with the Breakout Board V2? I tried to setup mine matching all the pin outs. Whenever I go for live view, the screen just says "Please wait.. ". I'm using Lepton 2.5.

VoodooBane, 2020-06-28:
I was able to get my Leptop 3.0  to work with https://store.groupgets.com/products/purethermal-breakout-board
You need to bend the pins at an angle and then remove the  Vsync and EN Pins "This new board has two new pins, VSync and Enable" I used 3mm double-sided tape on the back of the board then soldering the pins together.

samirakle, 2020-07-02:
> I was able to get my Leptop 3.0 to work with https://store.groupgets.com/products/purethermal-breakout-board
> You need to bend the pins at an angle and then remove the Vsync and EN Pins "This new [...]

Hi there. Do you know if that Pure Thermal breakout board will work on the Lepton 3.5? Also, for clarification, which pins did you solder together exactly? I'm assuming it was the 2 extra pins (VSYNC and EN)? Thanks.

VoodooBane, 2020-07-02:
That was my mistake I went all out as well and got the leptop 3.5. That is
correct I the 2 pins extra are VSYNC and E. just desolder those ones of.
yes been the pin and solder them. I will be back with a photo for you! and
ill link the tap I used.

samirakle, 2020-07-02:
@VoodooBane Okay thanks. Yes, a picture would be great to see. I'm currently working on this build but I was provided the FLIR breakout board V2 which has 2 rows of pins (total of 20 pins). The Pure Thermal breakout board should be on its way sometime soon. 👍🏼

VoodooBane, 2020-07-06:
The solder job:
[image]

And here is the Tape I used it on the battery and the pure thermal board to keep it in place soldering (I did cover the PCB and the back of the pure thermal board with Kapton take)
[image]

I hope this helps! I wish we could utilize the Vsync pin... The screen lags when moving :(

VoodooBane, 2020-07-06:
I forgot to mention you. @samirakle

samirakle, 2020-07-09:
@VoodooBane Thank you so much for posting a pic! I'll let you know if I have any issues.

Ailsamly, 2020-08-20:
> @VoodooBane Thank you so much for posting a pic! I'll let you know if I have any issues.

@samirakle @VoodooBane Hi friends, I am a new learner from Germany and I am also planning to build a small thermal camera and integrate to a mobile robot, if possible, can we get in touch with each other and have a discussion for some problems. I would be really appreciated if I can get your experienced suggestions in my initial period. Thank you very much! My email address is whumly1223@gmail.com. You are welcome to contact me!
Best,
Leyuan

closed 2021-03-10 by maxritter