maxritter/diy-thermocam #76: Thermal images problem

liaotianjiao, 2019-08-29:
There is a problem that confused me for a long time that Thermal images is really bad!!!When I turned on my device,the images shows on the screen can hardly recognize at all.There are many straight line refresh in high frequency and detected temperature is wrong too. I tested the Lepton alone,and it turn out to be good.so it's not the problem of device.And I find out that the images shows on the screen is zoom in? Because the feild of vision when I tested Lepton alone is different to DIY-Thermocam.
     Really appreciate your reply!!!  :)
[image]

GuiotV, 2021-01-06:
Hey i have the same issue did you find a solution to this ?

GuiotV, 2021-01-31:
Found it !!! The Sensor was pulled out of a Flir one Gen2 Android. And in the firmware the leptonhw was not detected and set to lepton2 without shutter which was wrong.  It just need one more if with the right leptonhw see https://github.com/maxritter/diythermocam_firmware/blob/40e9c8a774ce4707db48dc1ce6462cbe0044825f/src/hardware/lepton.cpp#L578

closed 2021-03-10 by maxritter