maxritter/diy-thermocam #78: lepton 2.5 without shutter

mikromel, 2020-06-11:
Hi there, 
I got my hands on a lepton 2.5, but with the shutter removed. Naturally, it does not work, but I'm not sure if because of the missing shutter, or mistake when building the thermocam.
I'm getting only "Please wait" when trying to show video from camera.

Can the missing shutter be worked-around to get it to work, or is the shutter indispensable?

Thank You

maxritter, 2020-06-14:
Hi @mikromel ,

the firmware of the DIY-Thermocam does a auto-detection of the installed Lepton version via I2C on startup. When the Lepton2.5 is inserted, it assumes the shutter to be present as well, b/c it needs it for radiometric functions.

What you can do is to trick it into assuming a Lepton2.0 Non-Shuttered is present. To do so, you have to modify the function "void lepton_version()" inside https://github.com/maxritter/diythermocam_firmware/blob/40e9c8a774ce4707db48dc1ce6462cbe0044825f/src/hardware/lepton.cpp, by setting leptonVersion = leptonVersion_2_0_noShutter; and 		leptonShutter = leptonShutter_none;.

Compile your own HEX as mentioned [here](https://github.com/maxritter/diythermocam_firmware) and flash it to the device. Let me know if this works for you :)

mikromel, 2020-06-15:
Hi Max,
thank You for the response. I did the change and after installing the spot sensor it works, except the normal camera.

I reports "cam not connected" . I'm sure the cam worked some time before, so I probably shorted it during the assembly yesterday. I've seen some sparks... 

Anyhow, thank You very much and kind regards

closed 2020-06-15 by mikromel