maxritter/diy-thermocam #79: KITs Available?

chick0n, 2020-07-08:
Hi

Is it possible to buy a Thermocam KIT without the  Lepton in Germany?

rjadawe, 2020-07-08:
Hello,If you can't find the complete kit, I have some extra components from my build.Primarily the main PC board as the minimum order was 3.The difficult part to find will be the lepton breakout board and lepton 3 or 3.5 sensors.Breakout board v1.4 is no longer available.I can look to see what other components I have left over if your interested.Thanks,Rob DaweSent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/maxritter/DIY-Thermocam/issues/79",
"url": "https://github.com/maxritter/DIY-Thermocam/issues/79",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

WoodsideOak, 2020-07-08:
I still have PCBs
[https://github.com/maxritter/DIY-Thermocam/issues/64](url)

chick0n, 2020-07-14:
Ok... PCB is on its way. :)

Maybe anyone has an "ArduCAM-M-2MP Plus" for sale in Europe?

Ailsamly, 2020-08-20:
> Hi
> 
> Is it possible to buy a Thermocam KIT without the Lepton in Germany?
> 
> Or anyone in Europe, has any parts left, I could buy?
> 
> I bought a broken flir one pro. I think this is the best option, to use the sensor if I cant repair it...

Hi, how are you going with your DIY project? I am a new learner from Berlin and I am also planning to build a thermal camera, if possible we can get in touch and have a discussion . My email address is whumly1223@gmail.com. 

Look forward to hearing from you.

Leyuan

Ailsamly, 2020-08-20:
> Hi
> 
> Is it possible to buy a Thermocam KIT without the Lepton in Germany?
> 
> Or anyone in Europe, has any parts left, I could buy?
> 
> I bought a broken flir one pro. I think this is the best option, to use the sensor if I cant repair it...

Hi, how are you going with your DIY project? I am a new learner from Berlin and I am also planning to build a thermal camera, if possible we can get in touch and have a discussion . My email address is whumly1223@gmail.com.

Look forward to hearing from you. @chick0n 

Leyuan

Ailsamly, 2020-08-20:
@rjadawe 
Hi friend, I am a new learner from Germany and I am also planning to build a small thermal camera and integrate to a mobile robot, if possible, can we get in touch with each other and have a discussion for some problems. I would be really appreciated if I can get your experienced suggestions in my initial period. Thank you very much! My email address is whumly1223@gmail.com. You are welcome to contact me!

Best,
Leyuan

maxritter, 2021-03-10:
Catching up on this, I just announced the DIY-Thermocam V3 on the mainpage. It will be available in 3-4 weeks and include a kit containing most of the components, that can be ordered on this website. Hope this helps :)

closed 2021-03-10 by maxritter