maxritter/diy-thermocam #80: Firmware Updater error

Juzmars, 2020-07-17:
Hi,

I have a problem concerning the Firmware Updater. I installed everything as mentioned in the documentation, but when I try to run the Updater.exe it doesn't respond properly and just displays an error message, that either the FlashReboot or the FlashHelper is missing. However both programs are in the same file as the Updater. I've tried several things so far, like running it in administrator mode or on a new account, but still the same issue. 

Thanks Michael

hoangminhanhtai, 2020-08-07:
Yes i also have the same problem with the newest firmware.
@maxritter : please take some times of yours to fix the firmware issues. Thanks and best luck to you

Ailsamly, 2020-08-20:
> Yes i also have the same problem with the newest firmware.
> @maxritter : please take some times of yours to fix the firmware issues. Thanks and best luck to you

Oh, so does it mean we cannot continue with this project if the firmware issue cannot be solved?

maxritter, 2020-08-30:
@Ailsamly @hoangminhanhtai @Juzmars Sorry for my late reply. For some people there are indeed issues with the official FW updater on Windows.

In this case, please use the Teensy Loader for Windows from here: https://github.com/maxritter/DIYThermocam/tree/master/Software/Firmware%20Updater/Teensy%20Loader

Load the hex file with "teensy.exe", then launch "teensy_reboot.exe" while the Thermocam is connected to the PC and turned on, the flash procedure should start. In case there is an error "File too big", upload "FileTooBig.hex" to the Teensy first and then the newest version of the firmware in the next step. In case there are missing dependencies, download and install Teensyduino from here first: https://www.pjrc.com/teensy/teensyduino.html

Please let me know if that worked :)

Ailsamly, 2020-08-31:
Hi Max,
I am a new lerner to this field and I am interested in your diy-thermocam.  But Can I directly get the image on the computer instead of using the display. If so, do I still need to print your PCB?

Thank you very much and I am looking forward to your reply.

Leyuan

| |
武汉大学-马乐原
|
|
邮箱：whumly1223@163.com
|

签名由 网易邮箱大师 定制

On 08/30/2020 11:58, Max Ritter wrote:

@Ailsamly@hoangminhanhtai@Juzmars Sorry for my late reply. For some people there are indeed issues with the official FW updater on Windows.

In this case, please use the Teensy Loader for Windows from here: https://github.com/maxritter/DIYThermocam/tree/master/Software/Firmware%20Updater/Teensy%20Loader

Load the hex file with "teensy.exe", then launch "teensy_reboot.exe" while the Thermocam is connected to the PC and turned on, the flash procedure should start. In case there is an error "File too big", upload "FileTooBig.hex" to the Teensy first and then the newest version of the firmware in the next step. In case there are missing dependencies, download and install Teensyduino from here first: https://www.pjrc.com/teensy/teensyduino.html

Please let me know if that worked :)

Juzmars, 2020-08-31:
> @Ailsamly @hoangminhanhtai @Juzmars Sorry for my late reply. For some people there are indeed issues with the official FW updater on Windows.
>
> In this case, please use the Teensy Loader for Windows [...]

Thanks, that solved the problem for me :)

maxritter, 2020-08-31:
You can use the thermal live viewer (https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Thermal%20Live%20Viewer) written in Python to directionaly stream to the PC and capture images there. But in any case, you need the PCB, as it connects the Lepton sensor to the MCU that is used for preprocessing and transfering the data to the PC.

Ailsamly, 2020-08-31:
Hi Max, thank you very much for your kind reply. But since I don't need the display, should I just leave the space for display in the PCB empty? Or maybe I can just connect lepton, breakout board and teensy on a breadboard? 

Leyuan
On 08/31/2020 16:09, Max Ritter wrote:

Hi Max, I am a new lerner to this field and I am interested in your diy-thermocam. But Can I directly get the image on the computer instead of using the display. If so, do I still need to print your PCB? Thank you very much and I am looking forward to your reply. Leyuan | | 武汉大学-马乐原 | | 邮箱：whumly1223@163.com | 签名由 网易邮箱大师 定制 On 08/30/2020 11:58, Max Ritter wrote: @Ailsamly@hoangminhanhtai@Juzmars Sorry for my late reply. For some people there are indeed issues with the official FW updater on Windows. In this case, please use the Teensy Loader for Windows from here: https://github.com/maxritter/DIYThermocam/tree/master/Software/Firmware%20Updater/Teensy%20Loader Load the hex file with "teensy.exe", then launch "teensy_reboot.exe" while the Thermocam is connected to the PC and turned on, the flash procedure should start. In case there is an error "File too big", upload "FileTooBig.hex" to the Teensy first and then the newest version of the firmware in the next step. In case there are missing dependencies, download and install Teensyduino from here first: https://www.pjrc.com/teensy/teensyduino.html Please let me know if that worked :)

You can use the thermal live viewer (https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Thermal%20Live%20Viewer) written in Python to directionaly stream to the PC and capture images there. But in any case, you need the PCB, as it connects the Lepton sensor to the MCU that is used for preprocessing and transfering the data to the PC.

maxritter, 2020-08-31:
@Ailsamly Yeah this is possible. The firmware will detect there is no display present, and directly switch to serial connection mode then.

Ailsamly, 2020-08-31:
Hi Max,
Thank you very much! Is it still possible if I want to put this thermocam on a drone and wirelessly get the image or vedio?

On 08/31/2020 16:40, Max Ritter wrote:

@Ailsamly Yeah this is possible. The firmware will detect there is no display present, and directly switch to serial connection mode then.

Ailsamly, 2020-09-03:
Hi Max,
Sorry to write to you again. But I have another issue regarding the PCB. Because the Flir lepton breakout board you used is not available now. What we can buy now is the breakout boardV2.0. So do we need to adjust the PCB?

Thank you very much and waiting for your reply.

Leyuan

On 08/31/2020 17:08, Ailsamly wrote:
Hi Max,
Thank you very much! Is it still possible if I want to put this thermocam on a drone and wirelessly get the image or vedio?

On 08/31/2020 16:40, Max Ritter wrote:

@Ailsamly Yeah this is possible. The firmware will detect there is no display present, and directly switch to serial connection mode then.

maxritter, 2021-03-10:
The new breakout board will be supported in V3, closing this issue now.

closed 2021-03-10 by maxritter