maxritter/diy-thermocam #82: Thermocam Freezes

chick0n, 2020-09-04:
And another Question:

Sometimes my Thermocam freezes at Bootup or the Picture Freezes,and it completely locks up, and I have to Switch on and off.

I also Charged the Lipo to 100%, and made Batterie gauge Calibration. But nonetheless it reports, that the Batterie is almost Empty.

What could be the problem?

I built the thermocam with an "ArduCAM-M-2MP Plus", and the new V2 Lepton Breakout board:
https://store.groupgets.com/products/purethermal-breakout-board

I connected like this:
[image]

Is there a way, to completely erase the teensy including EEPROM / Calibration Data, and start over?

chick0n, 2020-09-05:
I still have the Freeze issues. I think it has to do with the ArduCAM-M-2MP Plus.

When I deactivate the Arducam, it works.

With Arducam on, I get freezes during bootup and operation. 

The Camera produces "Garbage" Output too. After about 10sec warmup time, the Camera works correctly.

So either my Arducam is defect, or it has something to do with firmware implementation?

So I have to boot the thermocam with Arducam disabled, then enable the Arducam and wait 10sec for warmup. Then everything works Ok.

maxritter, 2020-09-06:
@chick0n Strange issues and hard to debug remotely.. It could be that the Arducam is faulty or the module received a HW upgrade that the firmware is not able to cope with..

If you can't fix the problem on your own, please drop me a mail to mail@maxritter.net, and we can see how we can handle the situation (sending the device to me for inspection f. ex.).

chick0n, 2020-09-07:
I ordered the Arducam directly from uctronics:
https://www.uctronics.com/camera-modules/camera-for-arduino/arducam-mini-module-camera-shield-w-2-mp-ov2640-for-arduino-uno-mega2560-board.html

Maybe the model is too "recent". :)

chick0n, 2020-09-09:
Its Hardware defect...when I heat up the Samsung RAM Chip with Hot-Air, it works.

So its Bad RAM Chip, or BGA Soldering Issue...

So maybe dont buy from uctronics....

chick0n, 2020-10-27:
Uctronics sent me an new Camera, for free...this one works OK so far...

ncbelov, 2021-01-25:
Need help! There is no image from the camera arducam 2mp plus-rev-A working camera checked on arduino uno. What needs to be changed in firmware 2.48 to make it work?

maxritter, 2021-03-10:
Closing the issue, glad it's working.

closed 2021-03-10 by maxritter