maxritter/diy-thermocam #83: Lepton 3.5

WolfZ, 2020-10-09:
Hi,
in the description it is stated the Lepton "including the radiometric versions (Lepton2.5 only at the moment)" are supported. Is this still correct or is Lepton 3.5 (the radiometric version) supported and the website slightly outdated.
Kind regards
Wolf

chick0n, 2020-10-14:
Yes...its fully supported.

2.45 (06.04.2018)
-Added support for radiometric Lepton3.5

I use a 3.5 too.

VoodooBane, 2020-10-15:
Really wish the Vsync pin could be used. The screen lags on regular
movement.

leoshmu, 2021-01-11:
@chick0n What breakout board did you use with the Lepton 3.5? The breakout board linked on the readme doesn't seem compatible with the 3.5

revnu, 2021-03-08:
Guys, could we please revive this project in some way? I was close to order all parts but stopped then I realized that the Breakoutboard is not produced any more. The Version 2 of the Lepton-Breakoutboard (250-0577-00 ) has 10 instead of 8 PINs, so it would not be compatible with the rest of the PCB-design I think. In other words that means that this poormans-thermocam is not available any more for "normal people", without re-building the circuit/casis.

Is there a chance we fork this in a way that part-availability is also take into consideration? Then this will be futureproof :)

revnu, 2021-03-08:
> 
> 
> @chick0n What breakout board did you use with the Lepton 3.5? The breakout board linked on the readme doesn't seem compatible with the 3.5

So maybe the Version2 of the breakout-board (250-0577-00) is supported? Is there any chance that the screwholes and the dimensions are identical so you could use the PCB. I saw, that the position of the sensor is slightly different, but that could be easy solved with another enclosure

maxritter, 2021-03-10:
I am working on a V3 of the Thermocam, that supports the new Lepton Breakout Board. Stay tuned ;)

helmarw, 2021-04-22:
cool project!
is the new Breakoutboard 2.0 already supported in the latest firmware 2.48 ?

maxritter, 2021-05-01:
@helmarw Thanks!
For the Breakout Board V2, you need PCB Version 3.0 and firmware version 3.0 from https://github.com/maxritter/DIY-Thermocam/releases/tag/3.0.0

helmarw, 2021-05-01:
oki, thnx!
is buydisplay.com the only shop where you can get this display? 
or are there other compatible displays, which might be easier to get ?!
also the lipo-charger seems to be only available on aliexpress
both will take weeks if not months to arrive ...

maxzev, 2022-01-12:
I had the same problem, just couldn't find a display with "parallel" interface like in the manual and bought the cheapest display with regular SPI interface instead.  It works so far. The display is 320x240 SPI v2 (exact model kmrtm32032-SPI with ILI9341 touch). It has its own pinout and doesn't match the PCB, I had to connect it just with wires.  I did no changes in the source code and it works ok, also lags from time to time but nothing critical.  Not sure if it bcs of the display.

cymatics, 2022-09-20:
After many weeks of waiting for parts, figuring out some of the less obvious details in the instructions, etc, I finally got my board fully assembled this week.  I installed the latest firmware, and booted up "Display", "Touch", "Battery Gauge", and "SD Card" all read "OK".  "Flir Lepton" came up  "failed".  After a few restarts to test, I have determined the Module works when using my mini USB hookup, and that when i start up the DIY cam board, the module's shutter immediately closes, and stays closed until I power down.

Using Lepton Module 3.5 with Breakout board V2.

maxritter, 2022-09-23:
@cymatics I have answered you via mail ;)

closed 2022-09-23 by maxritter