maxritter/diy-thermocam #84: I can't make the Lepton Breakout Board 2.0 to work

Lionear1508, 2020-11-09:
Hello everybody, 

I got my Lepton 3.5 from FLIR with the Breakout Board 2.0 (I tried to get the V1.4 but I think is obsolete and I found in this forum that somebody used this one: https://store.groupgets.com/products/purethermal-breakout-board but right now is sold out), I tried to make the connections like if I was connecting the V1.4 (I got the Datasheet of the V2.0), I solder some male-female jumpers to the PCB to can do some testing with this board but when I turn on the camera it just stuck in the logo and it says booting..... I was wondering if somebody can just help me with that or can give me an idea about what is happening.

I dont know if I explain my issues in a correct way or if this section is the right one, it's my first time in here, I apologize in advance.

I will thank you very much!!

VoodooBane, 2020-11-09:
I am the person that used the purethermal break out board. I do have a Breakout Board 2.0 as well.  But i would reccomend using dupont jumper wires to test before soldering the board onthe break out board the pins are pretty much Reversed. Use ur multimeter conductivity mode to make sure the pin out on the board is matching the Pins. Also to Note you prob have to solder bridge the jumpers on the back of the breakout 2.0 board to enable the pins. do not just snip pins make sure if the are using the jumpers that came with them then the conductivity mode testing showing the pins are actually going out to the 2.54mm pins. you what I am saying? The whole point the snipping the jumpers on the back for the board is so that the actual breakout baord and lay somewhat flush to the custom PCB. Also take you time and be patient. You do not want to ruin "my precious" lepton 3.5 (it is expensive and just sooooo tiny!!!! amazing!!!! Aka dont accidentally wire a data pin to a voltage pin. you know what i mean?Ask if you have any other questions and ill tryingto tinker with mine to see what i can do to help.

Lionear1508, 2020-11-11:
Hello @VoodooBane thanks for answer me!

I made it work yesterday, I saw the datasheet for the pinout and the pin number 2 is where I connected the voltage of my power supply, but nothing happened, so I found in the FLIR tutorials that you can use another two pins next to the lepton sensor for the power supply inputs, I just connected my 3.3V on it and the shutter just closed and opened! but still with no communication, I don't know why but I downloaded the program again (I did not modified anything) and it looks like it's working, maybe is my setup.

I really aprecciate that you take the time to answer.

closed 2020-11-11 by Lionear1508

VoodooBane, 2020-11-11:
Awesome glad to hear it!