maxritter/diy-thermocam #86: Arducam 2mp mini plus ov2640

ncbelov, 2021-01-25:
Need help! There is no image from the camera arducam 2mp plus-rev-A working camera checked on arduino uno. What needs to be changed in firmware 2.48 to make it work?
[image]
[image]

maxritter, 2021-03-08:
Hey @ncbelov,

sorry for the long waiting time. There have been repeated reports about issues with a newer hardware version of the Arducam and the DIY-Thermocam. I just ordered a new Arducam by myself and will let you know once I have a fix available.

Best,
Max

ncbelov, 2021-03-10:
The camera does not work on firmware 2.48 writes please wait. I am waiting for the updated firmware.

boriskogan81, 2021-04-07:
Same issue with the same component. I am also not sure how to assemble this physically, since the instructions deal with the previous version (UC-354)-am nervous about breaking it. Please advise.

closed 2023-05-06 by maxritter