maxritter/diy-thermocam #89: Cam hangs on Boot Screen, Hardware Issue?

Catarallus, 2021-04-22:
Hello together,

my Thermocam won't boot, what I see is the Boot Screen. I guess it's is a hardware problem.

I use the PCB Rev 2.3 with the new Lepton Breakboard V2.0 (20 Pins) and the Lepton 3.5. I also noticed that the Diode on the Lepton Breakoutboard is in the wrong direction. I wired the Voltage source to J3 connector to solve this Problem.
The Arducam is from robotshop which look's like a litte bit different like in the assembly guide, but the pin assignment is the same. In the new BOM is no Arducam listed, is this part obsolet?
I have no MLX90614 installed because in don't need it, right?

The voltage on the Lepton Breakboard and the Arducam is about 5V.

Any suggestions what else I can do to fix the Problem?

maxritter, 2021-05-01:
Hey @Catarallus!

For the Lepton Breakout Board V2, you need the PCB 3.0 and the firmware version 3.0.
Then the Lepton3.5 will work out-of-the-box and you do not need the MLX90614. 

Unfortunately the visual camera is not supported any more on V3, as it made some problems. But you can always use a regular camera and create combined images on the PC with the ThermoVision software.

Best,
Max

Catarallus, 2021-05-03:
Hi Max,

tank you for your fast reply.
Flashing Pre Compiled V3 failed with the Teensy Uploader. It seems the file can't be loaded, "unreadable" is shown in the Uploader Statusbar.

maxritter, 2021-05-05:
Hey @Catarallus,

do you use the Teensy 4.1 and the firmware loader from [here](https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Firmware%20Updater)? Then it should work, just tried it out myself.

Best,
Max

Catarallus, 2021-05-05:
Hi @maxritter,

yes I use Teensy 4.1. The flash process is now succesfully, but it only works with the push button (it is not easy to push the button if the Teensy is soldered to the PCB already) But who cares.
After Power up the screen is black, now. 
I use farther PCB Rev 2.3. The Lepton is contacted with some wires on the PCB, of course I payed attention to the correct Pins. (The Thermocam will be mounted on a drone, so it's not a Problem). 
What are differents beetwen PCB Rev 2.3 and PCB 3.0? Do I really need the new PCB?

Primergi, 2021-05-05:
Hi! Are you sure Breakout Board V2 works? You need to remove the Thermocam, and measure the DC-DC voltage.Must be Vcc 1.2 v, Vcc 2.8 v, Vcc 3.0 v.

Catarallus, 2021-05-06:
Hi,

I here the a "click" noise from the Lepton, it sounds like a camera shutter.
Which measurement points I have to check?

greetings
Johannes

Primergi, 2021-05-06:
"Which measurement points I have to check?"
Photo of the Breakout Board on the back side please.

Catarallus, 2021-05-07:
[image]
[image]

Primergi, 2021-05-07:
[image]

Primergi, 2021-05-07:
Sorry if something is unclear. I am Russian, and I use a translator :)

Jandroks, 2021-05-08:
From what you wrote it appears that you connected Teensy 4.1 to the PCB top and therefore nothing can work for you. I put together Thermocam v3. and it worked without any problems.

Catarallus, 2021-05-11:
@Jandroks 
thanks for your comment.
The pinout between Teensy 4.1 & 3.6 seems the same. Do you also know my combination is not working?

Catarallus, 2021-05-11:
@Primergi 
> Sorry if something is unclear. I am Russian, and I use a translator .

I understand your suggestion. I will do the measurement if I had time to go ahead with this project.

closed 2023-05-06 by maxritter