maxritter/diy-thermocam #91: MTP device Windows 10

helmarw, 2021-05-06:
hi,

i just got my teensy 4.1 and first wanted to try out loading the firmware.
so i plugged it in and loaded the v3 hex-file (using Teensy Loader 1.52 under macOS) no error.
then in unplugged it, put a 2GB SD card in (formatted with SD Formatter, FAT16)
and plugged it into a Windows 10 PC (since macOS doesnt handel MTP device
Device is being recognized but i cant access the filesystem ... (reinstalled MTP drive too...)
even if nothing is connected to the teensy (except the SD card) shouldn't i be able to access its filesystem at least ?
Best
Helmar

maxritter, 2021-05-06:
@helmarw The MTP mode is not activated at startup automatically. Instead when the device is turned on and a USB cable is inserted, a prompt on the screen asks the user whether he(she wants to transfer files. Only then the Thermocam switches to MTP mode. 
So I assume that everything related to the flash is working and also your SD card does, but you need the display attached to verify it.

helmarw, 2021-05-06:
ic, thnx a lot!
then ive to wait for the display :/

helmarw, 2021-05-07:
you dont happen to have a display left over ?

psarossy, 2021-07-27:
With the latest 3.0.2 firmware, I'm not getting the prompt on Windows 10 (or MacOS).

Verified that another MTP device (Sony camera) does show up as expected on Win10 with the same cable.

maxritter, 2021-07-31:
> With the latest 3.0.2 firmware, I'm not getting the prompt on Windows 10 (or MacOS).
> 
> Verified that another MTP device (Sony camera) does show up as expected on Win10 with the same cable.
> 
> [update] I checked again in the evening, the Teensy shows up on Win10 but nothing on the screen nor does it get attached.
> [image]

I just re-checked MTP functionality on 3.02, but it should work as expected:

[image]

You have to turn the camera on and have it in live mode. Then insert the USB cable and the USB File Transfer prompt appears on the screen. Choose "Yes" and afterwards you will see the "Teensy" device under your Computer. In case there is a driver missmatch, try to install this one: https://www.pjrc.com/teensy/serial_install.exe. Also make sure that the SD Card has 16GB or less and Class4 or Class10 (preferred). You can try to format the card on your PC first with a tool like SdFormatter: https://www.sdcard.org/downloads/formatter/

Let me know if this works for you :)

psarossy, 2021-07-31:
After fixing my power regulator issue, and unplugging and re-plugging the cord, eventually I got the pop-up, and was able to make the transfer work.

closed 2023-05-06 by maxritter