maxritter/diy-thermocam #92: Image transmission via wifi

Janusz-Janiec, 2021-05-14:
hi Max
I put together a Diy-Thermocam according to your design and it works fine. Have you ever wondered whether to use diy-thermocam to send an image via wifi. It would be an interesting option that would make the camera more attractive and find a new crowd of enthusiasts for your venture. I am asking because I am playing with the construction of drones and I thought that it would be worth using the camera to transmit the image in this way. There is a project that uses a version 2 camera and a Raspberry PI designed by you to transmit the image. What do you think.

maxritter, 2021-05-16:
@Janusz-Janiec Glad to hear the assembly worked and the Thermocam is alive!

Thanks for the suggestion. I am already thinking about a DIY-Thermodrone that has a much more compact form factor and can be attached to a drone for image transmission over WIFI, HDMI or analog video. Unfortunately I am not having much time right now for this as I want to focus on the delivery of the V3 and the extension of the software ecosystem there first, but be assured it's in the back of my mind ;)

Janusz-Janiec, 2021-05-17:
Thank you for your response.   Good luck in this and your future endeavors.

closed 2023-05-06 by maxritter