maxritter/diy-thermocam #93: finished my Themocam, a few remarks for the Enclosure

helmarw, 2021-05-14:
got my display finally :)
every thing is working fine !

[image]

i decided to 3D-print the enclosure first before i order a laser cut (if any)
good thing i did, since then i noticed few misalignments.

1) the opening for the USB-Port is to high
[image]

i also used a different t-Nut, because i could not get the one you suggested. only some minor modification nessesary.

2) the hole for the LED has to be 1-2mm more to the center or you need to leave the wires longer
[image]

3) most important, the hole for the Lepton is a bit low, i saw the upper edge of the housing in the thermal image
so i cut it a bit higher
[image]

4) i really didnt like the round head philips screws, so i used black A2 countersunk torx-screws. I think it gives it a nicer finish

helmarw, 2021-05-15:
i forked your project and added my modifications to my fork. Also added BOMs to all parts i bought
Feel free to use it, if you find it useful...

due to minimum order amount I also have 2 PCBs, 2 Lipo Chargers and a couple of T-nuts left over
if anyone is interested please drop me a line

maxritter, 2021-05-16:
@helmarw Great to hear it's working now!

The 3D printed enclosure looks great and thanks for the mentioned improvements. I have widened the Lepton cutout a bit to the top as you have suggested, and also slightly extended the microUSB port. Concerning the Charging LED, I am wondering if you have soldered it on the "right" side of the PCB. Maybe this was a bit clear in the first version of the assembly guide, I have updated the guide today with fresh images: https://www.diy-thermocam.net/build-your-own

helmarw, 2021-05-16:
hi, thnx for the reply
i checked again, i soldered the LED on the correct side of the pcb (other wise it would not hit the hole at all).
in the picture it looked like it needed to be as close as possible to the PCB, but in fact it needs a distance of 1-2mm.
so its better to mount  two or three sides of the housing to the PCB (on the display side) and then try inserting the LED
with 90deg bent leads. once its fits in the hole you can solder it.
in my case the leads are very short, almost flush with the PCB, thats how i interpreted your pictures. A higher res side view of the LED would have be helpful here, to see that the LED needs a bit space to the PCB
Now the LED sits at a aprox. 30deg angle inside the hole. its still working but, so i leave it as it is.

helmarw, 2021-05-18:
i updated my housing again: https://github.com/helmarw/DIY-Thermocam/tree/master/Images
i put all the stl und dxf files in my fork too

USB-Port fits nicely now
[image]

LED a bit better now, not perfect but close enough
[image]

front with an 8mm hole
[image]

and added a webcam-cover as lens protection :)  [simple 8mm one from Amazon](https://www.amazon.de/dp/B081J8D756/ref=cm_sw_em_r_mt_dp_3EE2QM3QVHKHH0B3Y5TW?_encoding=UTF8&psc=1)
[image]

IAowSwels, 2021-06-23:
Did you already covered these improvements in the latest CAD files?

closed 2023-05-06 by maxritter