maxritter/diy-thermocam #94: Issue with firmware.hex, Teensy.exe says it’s too large

salsa33, 2021-06-23:


salsa33, 2021-06-23:
Tried getting the firmware onto the Teensy by using VSCode, but now I'm getting errors with MTP even after unzipping the updated info to the correct folder.

maxritter, 2021-06-27:
Hey @salsa33,

did you assemble the DIY-Thermocam V2 or V3? 

Try to flash this hex file first: https://github.com/maxritter/DIY-Thermocam/blob/4b86ec988f69ddeae3e1c9218e800dde4bc6906f/Software/Firmware%20Updater/Teensy%20Loader/FileTooBig.hex, then try again the compiled firmware version. Make sure that if you have a V2 device, only flash firmware version starting with 2.x. 

Hope that helps!

salsa33, 2021-06-27:
The board has V3 printed on it. I'm not the one who built the kit, I just inherited the issue XD. I'll try this hex file and see what happens. I'll be working on it later today anyways. Thanks for the response!

salsa33, 2021-06-27:
Works now, I didn't even need the other hex file, this time when I opened teensy.exe it loaded just fine. Either way, thanks for responding.

closed 2021-06-27 by salsa33