maxritter/diy-thermocam #96: Step up converter overheating and powering down on battery

psarossy, 2021-07-27:
I've just received my Lepton 3.5 sensor, and finished the assembly.

The unit boots up, works for a few seconds, then the screen turns white and starts "blinking" in tandem with the red LED. Leaving the unit offline for some time, it works again for a few seconds, if I retry immediately it starts booting, it goes back to the loop immediately. 

This lead me to believe that there's a thermal issue with a component, and it seems that If I use the built in battery, the 5V step up converter overheats and shuts down at around 110C, then the board enters a reboot loop as the power rails are not stable.

Video: https://photos.app.goo.gl/JVPJjzPG8Rtgwq3A6

[image]

Thermal image
[image]
(can't get the raw file due to https://github.com/maxritter/DIY-Thermocam/issues/91#issuecomment-887789566)

maxritter, 2021-07-31:
Sorry to hear you are having an issue with the device. I haven't heard of this issue so far, all V3 customers have reported a working device since now. But it's always possible there is a faulty component.. 

- Have you verified with a multimeter that at step 3 of the [assembly guide](https://www.diy-thermocam.net/build-your-own), the two mentioned pins are separated? 
- Does the Lipo Charger work, e.g. what is the voltage that you measure on the Lipo battery after it is fully charged? 
- What is the voltage that you can measure between the middle pin and left/right pin of the Booster?

Let us debug this together, I hope we can get your device to work soon :)

psarossy, 2021-07-31:
Apparently this was a complete user error, apologies for wasting your time...

I did cut score the connection between the pads and remember measuring it, apparently I didn't score it deeply enough and there was a tiny, high resistance path still there, which was enough to mess with the regulator...

Scored it a few more times, and the resistance now is infinity \o/ and the regulator warms up but no longer overheats.

closed 2021-07-31 by psarossy