maxritter/diy-thermocam #98: Lepton sensor diagnostic failed

GautierR, 2022-03-03:
Hi,

Just finished the assembly of my camera. Unfortunately I have a Self-Diagnostic failed for the FLIR Sensor while booting. Is there a way to get a more descriptive issue ? I have double check all the soldering point of the FLIR board. 

I was thinking that it may be due to the sensor not being push enough, but I clearly hear a click. You have bellow some pictures. I got my sensor on Mouser. 

Any advices ? Thanks a lot

[image]

[image]

[image]

helmarw, 2022-03-03:
Do you know if the sensor is working at all? Did you test it e.g. on a raspberry first with standard code from flir?
just asking because my first Lepton i got from digikey was broken from the start. after some discussion with flir customer support and digikey i got a refund (not really a 5cent device, so it makes sense to give it some effort :) ). 
second one i bought from mouser and everything work fine from the start ...
I dont think you can do much wrong with this DIY Thermocam, i built it myself and using it ever since ...

GautierR, 2022-03-04:
I tried to build the code for the raspberrypi_video (see here : [LeptonModule/software/raspberrypi_video](https://github.com/groupgets/LeptonModule/tree/master/software/raspberrypi_video) ) without success.

I am using a raspbian dist, and not able to install qt4-dev-tools (needed for qmake). 

Anyway, I have ordered a new sensor to see if it can works just by switching them.

GautierR, 2022-03-04:
I have actually managed to get it compiled with qt5 and run the software.

It seems to works even if it's very slow and the refresh is not fast at all.

[image]

I have plugged all back to the DIY-Thermocam PCB, and still get the same issue during the boot

GautierR, 2022-05-27:
I bought a new sensor, solved the issue.

BTW, I have also designed an updated version of the 3D printed enclosure (thinner and more smooth). I will try to make another iteration to fix some issues and upload both STEP and STL if someone wants it.

[image]

[image]

zhaocundang, 2023-07-22:
IIC OK , SPI OK, but touch continue lcd show "lepton error , try to reset.." I don't  know where is wrong , how to deal with it