maxritter/diy-thermocam #101: Building Instructions v3 Feedback: LED, Lipo Protector, Switch, Heatshrink, Slide Cover

rduplain, 2022-12-01:
Hello! The thermocam, kit, and documentation are most excellent! After assembling a v3 kit, here are areas that could be improved in the documentation. Each topic is a simple suggestion to say emphasized sentence in the docs.

* **LED (14, 15).** _The LED needs to be soldered with enough lead wire to bend it 90 degrees upward and extend beyond the PCB._ I let gravity decide when soldering, attaching at the position in which the LED wire was already bent. The result was too short, which I only discovered after using a flush cutter, so I had to shave the LED and PCB in order to fully bend 90 degrees.
* **Lipo Protector (23).** _If desired, trim excess from the Lipo Protector with scissors before removing any protection foil. It just needs to be slightly bigger than the Lipo battery._ It's much more difficult to cut once the protection foil is removed, and the Adafruit lipo is smaller than the Samsung shown in the pictures.
* **Switch (30).** _Solder the wires perpendicular/horizontal to the switch, as there will be very little clearance once assembled, and ensure that enough insulation stays in tact to avoid shorting the two wires if they cross._ I needed to resolder these, and while they weren't touching, I used self-fusing silicone tape to ensure enough insulation.
* **Heatshrink (31).** _Make the heatshrink tubes as small as needed to cover the pads, as the button enclosure is a tight fit once the screen is added._ The heatshrink is only so flexible, and this put a lot of strain anytime I reseated the screen during the build, breaking the solder joint at the PCB multiple times until the heatshrink was broken in enough.

Ultimately, I found the slide cover ("Lepton Enclosure Protector") in the kit was not usable as it would result in a "vignette" of heat in the resulting image (i.e. the sensor was seeing the cover when either opened or closed). I reseated the cover in every position and orientation I could think of before finally giving up and discarding it.

The firmware instructions are a bit vague. The build instructions send you [here](https://github.com/maxritter/DIY-Thermocam/tree/master/Software/Firmware%20Updater) which doesn't make obvious what the steps are. This wasn't an issue as I've used Teensy before: (a) press the button the board to enter programming mode, (b) `teensy_loader_cli -w --mcu=TEENSY41 firmware.hex` (I'm using the Teensy CLI, not the GUI).

The build instructions could suggest not adding the plate (called "backside") until after firmware is flashed, because you'll have to take it off to access the button to put the microcontroller into programmer mode.

Altogether, the build instructions are great as is. I had no problem putting the kit together and starting to get thermal images!

Thanks!

PS: I don't see the source of https://www.diy-thermocam.net/building-instructions/ posted anywhere, else I would have submitted a patch.

maxritter, 2023-05-06:
Hey @rduplain,

sorry for the late reply, I have not been very active on this project for some months.

Thanks a lot for the detailed feedback on how to further improve the guide. I have included the changes in the latest version of the manual. The website now also has its own repository to give feedback: https://github.com/maxritter/diy-thermocam-website

closed 2023-05-06 by maxritter

rduplain, 2023-05-06:
Great, thank you! :rocket: